Kinovarobotics/ros_kortex #1: Meshes in urdf begin with ""package://JACO3_URDF_V10..." in the kortex_description package

AshisGhosh, 2019-01-25:
The urdf and meshes are located in the `kortex_description` package. However, the urdf specifies for the meshes to be loaded from a `JACO_URDF_V10` package.

thieraufc, 2019-02-25:
Hi @alexvannobel , are you able to offer a timeline on the release of 1.1.6? I would also be interested in having functioning launch files for the gen3 arm.

alexvannobel, 2019-02-27:
Hi @thieraufc !
The software package needed additional testing before release, it's why it was delayed.
Unfortunately, I can't provide you with a specific date as the release will depend on the test results.
As for the launch files, I am currently working on them! They will be uploaded with release 1.1.6.

alexvannobel, 2019-04-02:
Hi,

Version 1.1.6 was just released and the JACO_URDF_V10 package is now named kortex_description.
As for the launch files, they should be uploaded to the kinetic-devel branch in the upcoming days.

closed 2019-04-02 by alexvannobel