Kinovarobotics/ros_kortex #4: Missing "Frame.pb.h" when catkin_make

LongfeiProjects, 2019-01-30:
I suppose that "Frame.pb.h" is a generated file by protobuf. It is missing during the catkin_make process in the catkin workspace. The log message will be posted at the end.

Firstly I followed the procedure to setup "kortex" repository which is can be achieved via the link: https://github.com/Kinovarobotics/kortex. Everything goes on well except I have to comment out "add_subdirectory(300-BaseGen3_low_level_bypass)" due to the missing of example folder.

Then I followed the instruction given in "README.md" to install protobuf.  I checked out 3.5.1.1 on the web (https://github.com/protocolbuffers/protobuf/blob/master/src/README.md), then I finished the following steps without any problem:
`$ ./configure`
`$ make`
`$ make check`
`$ sudo make install`
`$ sudo ldconfig # refresh shared library cache`

Lastly, I git cloned the ros-kortex into catkin workspace, and simply run catkin_make without any modification. Then the compile failed. I am wondering whether there is a bug or **I missed some essential steps.**

message after running catkin_make:

> Base path: /home/longfei/catkin_ws
> Source space: /home/longfei/catkin_ws/src
> Build space: /home/longfei/catkin_ws/build
> Devel space: /home/longfei/catkin_ws/devel
> Install space: /home/longfei/catkin_ws/install
> ####
> #### Running command: "cmake /home/longfei/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/longfei/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/longfei/catkin_ws/install -G Unix Makefiles" in "/home/longfei/catkin_ws/build"
> ####
> -- The C compiler identification is GNU 5.4.0
> -- The CXX compiler identification is GNU 5.4.0
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Using CATKIN_DEVEL_PREFIX: /home/longfei/catkin_ws/devel
> -- Using CMAKE_PREFIX_PATH: /home/longfei/catkin_ws/devel;/opt/ros/kinetic
> -- This workspace overlays: /home/longfei/catkin_ws/devel;/opt/ros/kinetic
> -- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
> -- Using PYTHON_EXECUTABLE: /usr/bin/python
> -- Using Debian Python package layout
> -- Using empy: /usr/bin/empy
> -- Using CATKIN_ENABLE_TESTING: ON
> -- Call enable_testing()
> -- Using CATKIN_TEST_RESULTS_DIR: /home/longfei/catkin_ws/build/test_results
> -- Found gmock sources under '/usr/src/gmock': gmock will be built
> -- Looking for pthread.h
> -- Looking for pthread.h - found
> -- Looking for pthread_create
> -- Looking for pthread_create - not found
> -- Looking for pthread_create in pthreads
> -- Looking for pthread_create in pthreads - not found
> -- Looking for pthread_create in pthread
> -- Looking for pthread_create in pthread - found
> -- Found Threads: TRUE  
> -- Found gtest sources under '/usr/src/gmock': gtests will be built
> -- Using Python nosetests: /usr/bin/nosetests-2.7
> -- catkin 0.7.14
> -- BUILD_SHARED_LIBS is on
> -- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> -- ~~  traversing 5 packages in topological order:
> -- ~~  - kortex_actuator_driver
> -- ~~  - kortex_device_manager
> -- ~~  - kortex_driver
> -- ~~  - kortex_vision_config_driver
> -- ~~  - kortex_description
> -- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> -- +++ processing catkin package: 'kortex_actuator_driver'
[... 174 lines cut ...]
> In file included from /home/longfei/catkin_ws/src/ros_kortex/kortex_actuator_driver/src/actuatorcyclic_proto_converter.cpp:17:0:
> /home/longfei/catkin_ws/src/ros_kortex/kortex_actuator_driver/src/actuatorcyclic_proto_converter.h:28:22: fatal error: Frame.pb.h: No such file or directory
> compilation terminated.
> ros_kortex/kortex_driver/CMakeFiles/kortex_driver.dir/build.make:62: recipe for target 'ros_kortex/kortex_driver/CMakeFiles/kortex_driver.dir/src/base_proto_converter.cpp.o' failed
> make[2]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_driver.dir/src/base_proto_converter.cpp.o] Error 1
> make[2]: *** Waiting for unfinished jobs....
> ros_kortex/kortex_actuator_driver/CMakeFiles/kortex_actuator_driver.dir/build.make:88: recipe for target 'ros_kortex/kortex_actuator_driver/CMakeFiles/kortex_actuator_driver.dir/src/actuatorcyclic_proto_converter.cpp.o' failed
> make[2]: *** [ros_kortex/kortex_actuator_driver/CMakeFiles/kortex_actuator_driver.dir/src/actuatorcyclic_proto_converter.cpp.o] Error 1
> CMakeFiles/Makefile2:527: recipe for target 'ros_kortex/kortex_actuator_driver/CMakeFiles/kortex_actuator_driver.dir/all' failed
> make[1]: *** [ros_kortex/kortex_actuator_driver/CMakeFiles/kortex_actuator_driver.dir/all] Error 2
> In file included from /home/longfei/catkin_ws/src/ros_kortex/kortex_driver/src/base_ros_converter.cpp:17:0:
> /home/longfei/catkin_ws/src/ros_kortex/kortex_driver/src/base_ros_converter.h:28:22: fatal error: Frame.pb.h: No such file or directory
> compilation terminated.
> ros_kortex/kortex_driver/CMakeFiles/kortex_driver.dir/build.make:75: recipe for target 'ros_kortex/kortex_driver/CMakeFiles/kortex_driver.dir/src/base_ros_converter.cpp.o' failed
> make[2]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_driver.dir/src/base_ros_converter.cpp.o] Error 1
> In file included from /home/longfei/catkin_ws/src/ros_kortex/kortex_driver/src/basecyclic_proto_converter.cpp:17:0:
> /home/longfei/catkin_ws/src/ros_kortex/kortex_driver/src/basecyclic_proto_converter.h:28:22: fatal error: Frame.pb.h: No such file or directory
> compilation terminated.
> ros_kortex/kortex_driver/CMakeFiles/kortex_driver.dir/build.make:88: recipe for target 'ros_kortex/kortex_driver/CMakeFiles/kortex_driver.dir/src/basecyclic_proto_converter.cpp.o' failed
> make[2]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_driver.dir/src/basecyclic_proto_converter.cpp.o] Error 1
> CMakeFiles/Makefile2:10651: recipe for target 'ros_kortex/kortex_driver/CMakeFiles/kortex_driver.dir/all' failed
> make[1]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_driver.dir/all] Error 2
> Makefile:140: recipe for target 'all' failed
> make: *** [all] Error 2
> Invoking "make -j4 -l4" failed

jpcotekinova, 2019-01-30:
Hi Longfei,

We are about to release 1.1.6. 
Let us know if you still have issue with the new version.

--
JP

thieraufc, 2019-02-22:
azfarmiskam's fix solved this issue for me: by downloading the C++ Kortex API and moving the contents `include` and `lib` folders uncompressed from that download to the `include` and `lib` folders of `ros_kortex/kortex_api`, I was able to compile.

alexvannobel, 2019-04-02:
Hi, 
The [build.sh](https://github.com/Kinovarobotics/ros_kortex/blob/master/build.sh) script (new in release 1.1.6) downloads and unzips the API in the right folder. It should make build errors such as the ones listed here disappear.
If anyone stills encounters this error, feel free to reopen this issue!

closed 2019-04-02 by alexvannobel

Achllle, 2020-01-20:
This issue arrises because of a failed or incomplete download, which happens with the `build.sh` script @alexvannobel mentions as well. We run this script several times on our CI and it fails occasionally, with one of the following errors. You likely received one of these errors as well:

```
--2020-01-18 02:42:37--  https://drive.google.com/a/kinova.ca/uc?id=1ASbEsulf5cByru8Hy1oBZJyNDBa9H22C&export=download&confirm=
Resolving drive.google.com (drive.google.com)... 172.217.193.102, 172.217.193.100, 172.217.193.138, ...
Connecting to drive.google.com (drive.google.com)|172.217.193.102|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: ‘kortex_api-1.1.6.zip’
     0K ...                                                    43.8M=0s
2020-01-18 02:42:41 (43.8 MB/s) - ‘kortex_api-1.1.6.zip’ saved [3278]
Archive:  kortex_api-1.1.6.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of kortex_api-1.1.6.zip or
        kortex_api-1.1.6.zip.zip, and cannot find kortex_api-1.1.6.zip.ZIP, period.
ERROR while extracting the kortex api. code = 9
```

```
--2020-01-18 02:07:14--  https://drive.google.com/a/kinova.ca/uc?id=1ASbEsulf5cByru8Hy1oBZJyNDBa9H22C&export=download&confirm=
Resolving drive.google.com (drive.google.com)... 173.194.193.139, 173.194.193.100, 173.194.193.138, ...
Connecting to drive.google.com (drive.google.com)|173.194.193.139|:443... connected.
HTTP request sent, awaiting response... 500 Internal Server Error
2020-01-18 02:07:20 ERROR 500: Internal Server Error.
ERROR while fetching the kortex api. code = 8
```

Simply re-running the build will likely fix this. However this fails more than I'd like and so a fix from Kinova would be good here.