Kinovarobotics/ros_kortex #5: ROS message error while running C++ examples

plinioMoreno, 2019-03-13:
After installing ros_kortex, I was able to run the following examples properly:

- `GetControlLoopParameters`
- `GetSensorSettings`
- `ReadAllDevices`

But when I run the `SetControlLoopParameters`, I got the following messages:
- Server: `[ INFO] [1552494833.992110908]: KINOVA exception: 4`
- Client: `[ERROR] [1552494833.992377045]: Failed to call SetControlLoopParameters`

When I run the `PlayCartesian` and `PlayCartesianPosition`, I got the following messages:
- Server: `[ERROR] [1552494976.854756141]: client wants service /RefreshFeedback to have md5sum cc527e10b6cbeb93059adfab44147b77, but it has afdcac728d5cb28b98f9f7ec4a0fa269. Dropping connection.
`
- Client: `[ERROR] [1552494976.855100999]: Failed to retrieve the cyclic data.`

alexvannobel, 2019-03-15:
Hi @plinioMoreno ,

For ```SetControlLoopParameters```:
- What command did you run to start the example? What were the arguments?
- What nodes were running when you started the example? With what command did you start these nodes?
- ```SetControlLoopParameters``` is a call to the kortex_actuator_driver, which can be started either by specifying the IP address if you are directly connected to the actuator, or by specifying the IP address of the robot base and the device ID of the actuator if you are only connected to the base. I believe your error may be related to the kortex_actuator_driver maybe not having the right deviceID parameter. 
- Can you try to launch ```SetControlLoopParameters``` the same way as you tried running this one? Do you have the same error or does the example work fine?

For ```PlayCartesian``` and ```PlayCartesianPosition```:
- What command did you run to start the examples? What were the arguments?
- Were the kortex_actuator_driver **and** kortex_driver nodes running at the same time when you tried running it?
- Your error seem related to the ```RefreshFeedback``` function call failing. The ```kortex_actuator_driver``` **and** the ```kortex_driver``` have a service called ```RefreshFeedback```. I think the error may be due to the example calling the wrong version service. Can you test the example with only the ```kortex_driver``` node running?

Hope this helps,
Best,

Alex

akihikoy, 2019-04-17:
Hello, I experienced a similar error in running the PlayCartesian example, and found a solution.

I executed PlayCartesian according to the instruction in kortex_examples/readme.md, which says:

1. Run kortex_device_manager, and then kortex_actuator_driver.
1. Run: rosrun kortex_examples PlayCartesian

PlayCartesian did not work.

After PlayCartesian stopped, there was an error from kortex_actuator_driver:

`[ERROR] [1555498638.245085735]: client wants service /RefreshFeedback to have md5sum cc527e10b6cbeb93059adfab44147b77, but it has afdcac728d5cb28b98f9f7ec4a0fa269. Dropping connection.`

The reason seems to be: the RefreshFeedback service used in PlayCartesian (in example_play_cartesian.cpp, kortex_driver/RefreshFeedback is used) is different from the actual service /RefreshFeedback provided by kortex_actuator_driver (kortex_actuator_driver/RefreshFeedback).

So, I **activated kortex_driver instead of kortex_actuator_driver**.  Then PlayCartesian worked as expected.

1. Run kortex_device_manager, and then **kortex_driver**.
1. Run: rosrun kortex_examples PlayCartesian

I think that the instruction in _kortex_examples/readme.md would be incorrect_.

I also think that this confusion was due to the same service names (kortex_driver/RefreshFeedback and kortex_actuator_driver/RefreshFeedback).  I do not really understand the relation between kortex_driver and kortex_actuator_driver since I just started to use ros_kortex yesterday, but using the same names would be okay if kortex_actuator_driver is an alternative of kortex_driver (kortex_driver=100hz version and kortex_actuator_driver=1khz version, or something like that).

hlamontagne, 2019-04-17:
Hi @akihikoy ,

Thanks a lot for your comment, we really appreciate it. You are right, there is an error in the Readme file, in order to start the Cartesian control, you need the kortex_driver and not the kortex_actuator_driver. Basically, if you want to communicate with the robot, use kortex_driver but if for some reason, you need to communicate with an actuator directly. use the kortex_actuator_driver. I<m very sorry about this and this will be corrected very soon.

Thank you again,

Hugo

akihikoy, 2019-04-17:
Hi @hlamontagne,

Thank you so much for the explanation.  Now I understand the roles of the kortex_driver and the kortex_actuator_driver more clearly.

Very best,

Akihiko

hlamontagne, 2019-04-17:
My pleasure. :)

plinioMoreno, 2019-04-18:
> Hi @plinioMoreno ,
>
> For `SetControlLoopParameters`:
>
>     * What command did you run to start the example? What were the arguments?
>
>     * What nodes were running when you started the example? With [...]

About the `SetControlLoopParameters` I will let you know in the coming week.
> For `PlayCartesian` and `PlayCartesianPosition`:
>
>     * What command did you run to start the examples? What were the arguments?
>
>     * Were the kortex_actuator_driver **and** kortex_driver nodes [...]

About the `PlayCartesian` example, I was able to run it by following your suggestion of running `kortex_driver`, also suggested by @akihikoy.

alexvannobel, 2019-04-23:
Hi @plinioMoreno ,

Have you been able to test the ```SetControlLoopParameters``` example by running the ```kortex_actuator_driver``` node at the same time as the example?

Cheers,
Alex

alexvannobel, 2019-08-13:
Hi, 
I will close this issue as this is no longer possible with version 2.0.0, because all services and topics go through the `kortex_driver` package now.

Best,
Alex

closed 2019-08-13 by alexvannobel