Kinovarobotics/ros_kortex #10: Missing head file, catkin_make failed

kracon7, 2019-03-21:
I'm using Ubuntu 16.04 and ROS kinetic

First, I followed the procedure to setup "kortex" repository from: https://github.com/Kinovarobotics/kortex.
I followed the following directory hierarchies:
```sh
examples/kortex_api  
┬  
├ include/
├ lib/  
└   ┬  
    ├ debug/  
    └ release/ 
```

Here's what I got after cmake installation:
```sh
jc@jc-Precision-5510:/opt/kortex-master/api_cpp/examples/build$ sudo cmake .. -DCMAKE_BUILD_TYPE=release
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
---------------------------------------------------------------------------------
Project 'Getting_started' args: 
Project building for: 'Linux' in 'release' mode
create TARGET_EXE_NAME: '01_api_creation'
create TARGET_EXE_NAME: '02-protobuf_object_manipulation'
create TARGET_EXE_NAME: '03_api_mechanism'
create TARGET_EXE_NAME: '04_error_management'
create TARGET_EXE_NAME: '05_notification'
---------------------------------------------------------------------------------
Project 'Getting_started' args: 
Project building for: 'Linux' in 'release' mode
create TARGET_EXE_NAME: '01_devices_routing'
---------------------------------------------------------------------------------
Project 'Getting_started' args: 
Project building for: 'Linux' in 'release' mode
create TARGET_EXE_NAME: '01-move_angular_and_cartesian'
create TARGET_EXE_NAME: '02-sequence'
---------------------------------------------------------------------------------
Project 'Getting_started' args: 
Project building for: 'Linux' in 'release' mode
create TARGET_EXE_NAME: '01-low_level_cyclic'
CMake Error at CMakeLists.txt:29 (add_subdirectory):
  add_subdirectory given source "300-BaseGen3_low_level_bypass" which is not
  an existing directory.

---------------------------------------------------------------------------------
Project 'Getting_started' args: 
Project building for: 'Linux' in 'release' mode
create TARGET_EXE_NAME: '01_vision'
-- Configuring incomplete, errors occurred!
See also "/opt/kortex-master/api_cpp/examples/build/CMakeFiles/CMakeOutput.log".
jc@jc-Precision-5510:/opt/kortex-master/api_cpp/examples/build$ make
make: *** No targets specified and no makefile found.  Stop.
jc@jc-Precision-5510:/opt/kortex-master/api_cpp/examples/build$ sudo make
make: *** No targets specified and no makefile found.  Stop.

```

Then I followed the instruction to install protobuf. No problem found expect for the same problem as [issue 5315](https://github.com/protocolbuffers/protobuf/issues/5315). Installation was successful after removing -Werror from src/Makefile
[... 148 lines cut ...]
[ 22%] Generating Python code from SRV kortex_device_manager/SetSafetyConfiguration
[ 22%] Generating Python code from SRV kortex_device_manager/SetSafetyWarningThreshold
[ 23%] Generating Python code from SRV kortex_device_manager/SetSafetyErrorThreshold
[ 23%] Generating Python code from SRV kortex_device_manager/GetPartNumber
[ 23%] Generating Javascript code from kortex_device_manager/GetModelNumber.srv
[ 23%] Generating Python code from SRV kortex_device_manager/GetPowerOnSelfTestResult
[ 23%] Generating Javascript code from kortex_device_manager/GetDeviceType.srv
[ 23%] Generating Javascript code from kortex_device_manager/GetBootloaderVersion.srv
[ 24%] Generating Javascript code from kortex_device_manager/SetMACAddress.srv
[ 24%] Generating Javascript code from kortex_device_manager/SetSerialNumber.srv
[ 24%] Generating Javascript code from kortex_device_manager/SetModelNumber.srv
[ 24%] Generating Javascript code from kortex_device_manager/GetSafetyEnable.srv
[ 24%] Generating Javascript code from kortex_device_manager/SetSafetyConfiguration.srv
[ 24%] Generating Javascript code from kortex_device_manager/SetSafetyWarningThreshold.srv
[ 24%] Generating Javascript code from kortex_device_manager/SetSafetyErrorThreshold.srv
[ 24%] Generating Python msg __init__.py for kortex_device_manager
[ 24%] Generating Javascript code from kortex_device_manager/GetPartNumber.srv
[ 24%] Generating Javascript code from kortex_device_manager/GetPowerOnSelfTestResult.srv
[ 24%] Generating Python srv __init__.py for kortex_device_manager
[ 24%] Built target kortex_device_manager_generate_messages_nodejs
[ 24%] Built target kortex_device_manager_generate_messages_py
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

sbrisebois, 2019-03-21:
Hi kracon7,

This is a known issue and it's in the pipeline for the fix.

The problem you have is caused by an error in the example files. 
You have to modify the file api_cpp/examples/CMakeLists.txt : 

Add a # on the line : 
add_subdirectory(300-BaseGen3_low_level_bypass)

To have : 
#add_subdirectory(300-BaseGen3_low_level_bypass)

Then empty the build folder and re-try the cmake command.

Best,
Sylvain

alexvannobel, 2019-03-21:
Hi @kracon7 ,

For your catkin_make problem:

- I think you may be missing a pre-required step : the kortex api has to be downloaded and copied in the src/kortex_api folder as explained in this [readme](https://github.com/Kinovarobotics/ros_kortex/blob/master/kortex_api/readme.md).
- There seems to be a bug in the kortex_examples [CMakeLists.txt](https://github.com/Kinovarobotics/ros_kortex/blob/master/kortex_examples/CMakeLists.txt) where the executables don't specify the dependencies to the drivers, so they build before the . I'll open a PR to fix this, it should be straightforward.

Hope this helps,
Best,

Alex

alexvannobel, 2019-03-21:
#11

kracon7, 2019-03-21:
Hi @sbrisebois 
This solved the cmake issue for kortex-api. Thanks!

alexvannobel, 2019-03-21:
Hi @kracon7 ,

I just merged the PR so catkin_make should now build the packages just fine if you downloaded the API and put it in the right place. Can you test if the fix worked?

kracon7, 2019-03-21:
Hi @alexvannobel 
I put the downloaded api in kortex_api folder and removed /kortex_api/include/google/
```sh
jc@jc-Precision-5510:~/catkin_ws/src/ros_kortex/kortex_api$ ls
include  lib  readme.md
jc@jc-Precision-5510:~/catkin_ws/src/ros_kortex/kortex_api$ ls include/
client  client_stubs  common  messages

```

The error message is too long so I put it in a [google doc](https://docs.google.com/document/d/1X-jjZydtA8wVIjjgMRtKPzzB3KS1d5a17vTUWRTCgQY/edit?usp=sharing)

alexvannobel, 2019-03-21:
I just did a fresh checkout of the master branch, I downloaded the kortex_api from the Google Drive link in the readme and put it in the right place. I also made sure to remove ```build/``` and ```devel/``` folders and ran catkin_make and everything built just fine. 
Your errors seem related to Protobuf. 
- What is the output of ```protoc --version``` in a terminal window?
- Did you ```rm -rf build/ devel/``` before re-running ```catkin_make```?
- Could you try reinstalling it?

hlamontagne, 2019-03-21:
Hi @kracon7 Canyou verify your protobuf version?
protoc --version
You absolutely need to use the 3.5.1

kracon7, 2019-03-22:
@hlamontagne 
`protoc --version` gives `libprotoc 3.7.0`

@alexvannobel 
I did remove build and devel before catkin_make.
I will reinstall protobuf 3.5.1 and try again.

kracon7, 2019-03-22:
@sbrisebois @alexvannobel @hlamontagne 
I did `git checkout 860bd12` to swithc to branch 3.5.1.1 and everything else the same as the instruction.
`catkin_make` works now. Thanks!

alexvannobel, 2019-03-22:
Glad it worked!

closed 2019-03-22 by alexvannobel