Kinovarobotics/ros_kortex #19: end-effector orientation

evankrause, 2019-04-26:
When setting the orientation of the end-effector (theta-x, theta-y, theta-z), my assumption was that these values are setting the (ya, pitch, roll) relative to the end-effector reference frame. However, that's not what I'm seeing when actually setting those values via ROS.  For instance, change theta-z seems to always rotates around the base-frame z-axis regardless of the end-effector orientation, but setting theta-x seems to rotate around the end-effector's x-axis. 

It's unclear if this is a bug, or if I'm just confused about how to use the (theta-x, theta-y, theta-z) values.

alexvannobel, 2019-04-26:
Hi @evankrause ,

The arm is in "Mixed" reference frame as of SWU 1.1.6 : the translations are in the base reference frame and the orientations are in the end-effector reference frame.

What ROS service are you calling to set the orientation of the end-effector and what are the arguments?
```PlayCartesianTrajectoryOrientation```?
We could send the same ROS service call to replicate the behavior you see.

Cheers,
Alex

evankrause, 2019-04-26:
Hi Alex,

Thanks for the quick reply. Yes, I'm using PlayCartesianTrajectoryOrientation. With the arm already in some orientation, say (theta-x,theta-y,theta-z) = (110,25,130), using PlayCartesianTrajectoryOrientation to set a new orientation to (110,25,140) for instance doesn't produce a pure roll around the end-effector z-axis as I would expect.

But, going from (110,25,130) to (120,25,130) does produce a pure pitch rotation around the end-effector x-axis, as I would expect.

For these service calls I'm using an orientation speed of 15, with no specified duration or additional constraints.

Thanks,
Evan

evankrause, 2019-04-26:
Actually, I think I figured out my confusion. The theta values are the rotations required to transform the base reference frame into the orientation of the end-effector reference frame.

closed 2019-04-29 by evankrause