Kinovarobotics/ros_kortex #20: Implement ros_control with several hardware interfaces

sjhansen3, 2019-04-30:
Several thoughts here:

It would be nice if you guys developed and maintained a ros_control package for use with moveit and others.

Franka did quite a nice job with their control interface - I'd take a look at it for reference. 
https://github.com/frankaemika/franka_ros

We have developed one ourselves but it has several shortcomings as of yet. It would be nice for Kinova to use the baked in ros_control feature to switch controllers when switching into impedance mode for example when the buttons on the end effector are pressed. It would also be super helpful because it touches so many components that Kinova maintains - loop rates etc.

It would be ideal for several interfaces to be exposed in this way.  @nsaif-kinova tagging you to take a look at this. It is not super high priority for Peanut atm since we wrote our own - but we would really prefer for Kinova to take this on from a maintenance perspective.

There are also consultants out there that specialize in developing these things I would recommend working with https://picknik.ai/ - they are maintainers on both ros and ros_control and I believe they have built many ros_control interfaces for robotics companies before.

Achllle, 2019-05-01:
Our implementation: https://github.com/peanut-robotics/ros_kortex/pull/2

alexvannobel, 2019-05-03:
Hi guys,

First off, thanks for sharing your work! It's greatly appreciated. 

For the upcoming release (coming somewhere around June), we focus on MoveIt! and Gazebo integration for the Gen3 arm and the Robotiq gripper, but ROS Control integration is clearly the next ROS feature we will be working on, for the next release (somewhere in fall). As it is mostly a ROS-specific feature (its development should not be impacted a lot by the firmware development), the development will probably take place directly on Github, also.  

Thanks a lot for the feedback and references!

Cheers,
Alex

phuicy, 2020-02-03:
Any update on this?

felixmaisonneuve, 2021-06-14:
Hi everyone,

There is still no ros_control package

We will keep this issue updated with further progress.

Best,
Felix

madan96, 2023-11-20:
Hello! We just released the hardware interface for people who are interested in using `ros_control` [https://github.com/empriselab/kortex_hardware](https://github.com/empriselab/kortex_hardware).

martinleroux, 2024-09-11:
Thank you for sharing your work.

I will be closing this issue as Won't do on Kinova's side since our future development will be focused on our ROS2 repository.