Kinovarobotics/ros_kortex #22: ros_control and/or moveiit integration

evankrause, 2019-05-02:
I'm curious about the timeline for releasing either ros_control or MoveIt code. Any thoughts on when this might be available?

Thanks,
Evan

alexvannobel, 2019-05-03:
Hi Evan,

We currently are working on Gazebo and MoveIt! integration coming up for the next release (somewhere around June). The ROS control integration will be the next ROS milestone after that. 
I invite you to take a look at #20 if you're interested in a third-party ROS Control implementation for the Gen3 arm.

Thanks a lot for the feedback,
Cheers,
Alex

evankrause, 2019-05-13:
Great, thanks for the update Alex. I think I'll hold off until the next release with MoveIt! integration.

closed 2019-05-13 by evankrause