Kinovarobotics/ros_kortex #23: Dependency on specific Protobuf version

Achllle, 2019-05-07:
The installation instructions specify `git clone https://github.com/protocolbuffers/protobuf --branch 3.5.1.1 (you must use this specific version)`. This works but requires a long installation/build on new machines, including on CI. It takes about 20 minutes to build, which doubles our build times and we run it tens of times each day.

Is it possible to switch to a binary installation to greatly speed up this process somehow? What makes it that this specific commit needs to be used and not a main release?

Achllle, 2019-05-07:
Duplicate of https://github.com/Kinovarobotics/kortex/issues/15
feel free to close whichever

alexvannobel, 2019-05-09:
Hi @Achllle ,

As for the newest release of kortex_api (1.1.6), we managed to combine the kortex_api and the protobuf library in one static library file. I modified the CMakeLists and build script to keep the google headers folder and to include it, so you will be able to ```catkin_make``` without Protobuf being installed on your machine.
However, if you want to use the [generate_all.sh](https://github.com/Kinovarobotics/ros_kortex/blob/master/generate_all.sh) script to regenerate the message (which uses ```protoc```), service and converter files, you will need Protobuf installed, but I don't think this will be used a lot; it is more of a developers' tool when we need to regenerate the package for new API's.

As for the Protobuf version we are using, we definitely cannot change it on the short term. 
I was told we chose this specific version because earlier versions did not have some features we needed for the API, but I'll make some tests to see if a more recent released version (>3.5.1.1) could work and what would be the impacts of changing the Protobuf version we use. 

I'll keep this issue opened to communicate any Protobuf-related updates I have. 

Thanks a lot for the feedback,
Cheers,
Alex

truhoang-ubtrobot, 2019-10-31:
Hello, 

I installed the protobuf v3.5.1.1 and when I `catkin_make` my workspace I get the following errors

```
In file included from /usr/include/gazebo-9/gazebo/msgs/MessageTypes.hh:66:0,
                 from /usr/include/gazebo-9/gazebo/msgs/msgs.hh:35,
                 from /usr/include/gazebo-9/gazebo/gazebo_core.hh:20,
                 from /usr/include/gazebo-9/gazebo/gazebo.hh:20,
                 from /home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/third_party/roboticsgroup_gazebo_plugins/include/roboticsgroup_gazebo_plugins/mimic_joint_plugin.h:37,
                 from /home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/third_party/roboticsgroup_gazebo_plugins/src/mimic_joint_plugin.cpp:23:
/usr/include/gazebo-9/gazebo/msgs/magnetometer.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^~~~~
/usr/include/gazebo-9/gazebo/msgs/magnetometer.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^~~~~
/usr/include/gazebo-9/gazebo/msgs/magnetometer.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
  ^~~~~
In file included from /usr/include/gazebo-9/gazebo/msgs/MessageTypes.hh:53:0,
                 from /usr/include/gazebo-9/gazebo/msgs/msgs.hh:35,
                 from /usr/include/gazebo-9/gazebo/gazebo_core.hh:20,
                 from /usr/include/gazebo-9/gazebo/gazebo.hh:20,
                 from /home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/third_party/roboticsgroup_gazebo_plugins/include/roboticsgroup_gazebo_plugins/disable_link_plugin.h:34,
                 from /home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/third_party/roboticsgroup_gazebo_plugins/src/disable_link_plugin.cpp:23:
/usr/include/gazebo-9/gazebo/msgs/joint_wrench_stamped.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^~~~~
/usr/include/gazebo-9/gazebo/msgs/joint_wrench_stamped.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^~~~~
/usr/include/gazebo-9/gazebo/msgs/joint_wrench_stamped.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
  ^~~~~
```

When I uninstall protobuf, the package builds fine.

My call to `protoc --version` is:
```
libprotoc 3.5.1
```

alexvannobel, 2019-11-04:
Hi @truhoang-ubtrobot ,

Since #24 and #25 , the Protobuf installation is optional and only necessary to auto-generate the ROS .cpp, .srv, .msg and .h files from the Kortex API, files we provide in the repository. As for the Kortex API, the specific Protobuf version the library uses is bundled with the Kortex API static lib. You don't need to install Protobuf to use the Kortex API, or the ROS driver.
The errors you get are because there is a mismatch between the Protobuf version Gazebo uses (Gazebo ROS plugins in fact) and the Protobuf version we use.
I'll close this issue also, as Protobuf is no longer required to install to use the driver.

Best,
Alex

closed 2019-11-04 by alexvannobel

truhoang-ubtrobot, 2019-11-04:
Hi @alexvannobel ,

Sorry for keep posting on this closed issue but since it's related I didn't want to open a new issue. I'm having version collision issue during build because as you said, gazebo uses the different protobuf version. And to fix that, I installed protobuf2.6.1 and that issue is fixed. However, when I rebuild the entire workspace, the `.proto` files from kortex_api causes build error because they were generated by protobuf3.5.1.

Any suggestion to overcome this would be much appreciated. Thanks.

alexvannobel, 2019-11-04:
Hi @truhoang-ubtrobot ,

Building from a clean install of ROS works for me, and from the latest commit of the `kinetic` branch (which also works on Melodic). I don't need to install Protobuf either, as Gazebo comes with the necessary files I believe. I also don't believe Gazebo9 needs Protobuf 2.6.1, as 2.6.1 is for Gazebo7.
A couple questions that might guide you:
- Are you on the latest commit from the `kinetic` branch?
- Have you followed the [installation instructions](https://github.com/Kinovarobotics/ros_kortex#build)? Did you get an error message?
- What does the console show when you build?
- If you do a clean install of ROS Melodic (I'm guessing you use Melodic since you use Gazebo9), do you still have errors?

Hope this helps,
Best,
Alex

truhoang-ubtrobot, 2019-11-06:
Hi @alexvannobel,

I removed my protobuf installation and `rm -rf build/ devel/` and rebuild and it works now. Thanks!

Best,

Tru