Kinovarobotics/ros_kortex #26: Cartesian reference frame

akihikoy, 2019-05-10:
Hello,

I want to know which frame is the reference of the Cartesian (end effector) pose.

We can compute an end-effector pose from the current joint angles and the robot description given by the URDF.  I used KDL to compute this.  The frames can be clearly defined; the base frame is base_link, and the end-effector frame is EndEffector_Link.

On the other hand, we can also get an end-effector pose via RefreshFeedback service, which contains output.base.tool_pose_* values.
However I do not understand on which frame these values are defined.

For joint angles `q = [ 6.0369  1.0921  3.3023  4.6336  6.0416  1.182   1.5333]`,
KDL: `x = [ 0.6592  0.0651  0.1971  0.5006  0.4804  0.4949  0.5232]`
RefreshFeedback: `x = [ 0.779   0.0593  0.2015  0.5006  0.4804  0.4949  0.5232]`
(Note: RefreshFeedback gives the orientation in Euler angles in degrees, which were converted to quaternion)

The orientations are the same, while there is small difference in the positions.  The reason would be the reference frame is different.
Then which frame is the reference of RefreshFeedback?

I think this is an important question since such a reference frame should be commonly used in other services such as PlayCartesianTrajectory.

Thanks!
--Akihiko

alexvannobel, 2019-05-14:
Hi @akihikoy ,

Do you have a Robotiq gripper installed on your arm? On boot-up, the robot detects whether or not the gripper is connected and sets the tool offset (not exposed by the API as of 1.1.7, but will be as of the next release) accordingly. 
If the gripper is currently connected on your arm, can you try the same experiment without the gripper? Are the results the same?

Cheers,
Alex

akihikoy, 2019-11-25:
Although the version is updated to 2.0.0, this issue/question still exists.

Now the `RefreshFeedback` service will correspond with `base.tool_pose_*` elements of the `/my_gen3/base_feedback` topic.

I changed the gripper setting from the web interface; `End Effector Type = Not Installed`.
But it did not solve the issue.

For a better understanding, I visualized the two Cartesian poses;
Red cube: pose computed by KDL (`base_link='base_link', end_link='end_effector_link'`) with the current joint status.
Blue cube: pose obtained from `base.tool_pose_*` of the `/gen3a/base_feedback` topic.
Video: 
https://youtu.be/hIsr0LnbNGg

The robot is operated by the wrist-teaching button for translation, and the pose controller of the web interface for rotation.
Since during the rotation, the blue cube does not move, the Cartesian controller of the web interface looks using the tool pose.

My question is simple: what is the tool pose? and where is it defined?

alexvannobel, 2019-12-02:
Hi @akihikoy ,

I believe there is a discrepancy between the URDF file and definition of the TCP inside the arm and the URDF file and TCP used by ROS (and MoveIt). I think this is why KDL and the base_feedback don't have the same values.
I'll talk to the Control team about the TCP definition inside the arm to see what are the changes we need to make in the ROS URDF, but I think it will only be a matter of adapting a couple reference frame offsets.
 
I'll update you soon on this issue.
Best,
Alex

juntuck, 2020-01-08:
Hi @akihikoy ,
Could your issue be related to mine? #60

alexvannobel, 2020-06-09:
Hi @akihikoy and @juntuck ,

The arm's firmware has some hardcoded values for end-effectors and tool frames. I added those frames to the xacro's and the tool_frame link should now correctly reflect the robot's feedback. It was also added to the arm move_group in MoveIt for every arm type.

I'll close the issue now but don't hesitate to reopen if you have questions!
Best,
Alex

closed 2020-06-09 by alexvannobel

elevenjiang1, 2021-02-06:
> Hi @akihikoy and @juntuck ,
>
> The arm's firmware has some hardcoded values for end-effectors and tool frames. I added those frames to the xacro's and the tool_frame link should now correctly reflect [...]

Hello,I want to know how to change this value of xacro file.By the way ,I want to know which is accuracy,xacro or base_feedback ?
This image the left is calculcate by KDL and right one is from the feedback.
[image]