Kinovarobotics/ros_kortex #27: Trajectory control

akihikoy, 2019-05-12:
I would like to report an issue of trajectory control.

There are following services:

- PlayCartesianTrajectory (example use: https://github.com/Kinovarobotics/ros_kortex/blob/master/kortex_examples/python/PlayCartesian.py )
- PlayJointTrajectory

However both are **not a trajectory control**, but a point-to-point control.
In other words, they are not a way to generate a motion from multiple via-points.

In ROS, sometimes SimpleActionServer of actionlib is used to provide a trajectory control, which can be accessed through the /follow_joint_trajectory topic.  It is commonly used, for example, in Universal Robots UR and Yaskawa Motoman.
In these robots, /follow_joint_trajectory provides a trajectory control that can traverse multiple via-points.  Each via-point consists of time-from-start, joint angles, and joint angular velocities.

Note that executing PlayJointTrajectory and PlayCartesianTrajectory multiple times does not offer the same functionality, since the motion stops after each execution of PlayJointTrajectory and PlayCartesianTrajectory; i.e. the joint angular velocities at each via-point are zero.

Thus,
1. The PlayCartesianTrajectory service in ros_kortex is confusing naming.  It should be MoveToCartesianPose or something like that.
1. How can we generate a motion from multiple via-points?  I could not find such a service in ros_cortex.  Is that not implemented yet?

akihikoy, 2019-05-12:
Since this is a common issue of (pure) kortex, I have wrote a similar [issue report in kortex](https://github.com/Kinovarobotics/kortex/issues/16).

alexvannobel, 2019-05-14:
Hi @akihikoy ,

Since this issue is duplicated in the [issue you opened in kortex](https://github.com/Kinovarobotics/kortex/issues/16), I will close this one and answer in the kortex one. It's mainly a naming problem that has to do with pure kortex, and the support of via-points, which also has to do mainly with pure kortex.
I can say, however, that the **follow_joint_trajectory** action server will be available in the next release for compatibility with MoveIt!, and will use the new Kortex RPC to send multiple via-points trajectories to the arm. 

Don't hesitate to reopen this issue if you have a ROS-specific question concerning trajectory control. 

Cheers, 
Alex

closed 2019-05-14 by alexvannobel

akihikoy, 2019-11-22:
Hi, although there is a new release, the list of examples of `ros_kortex` seems not including `follow_joint_trajectory`.  Is that available now?
If it is available, could you explain how to use it?

alexvannobel, 2019-11-25:
Hi @akihikoy ,

As we don't have a custom motion planner, we interfaced MoveIt with our action server to send trajectories to the arm.
To visualise the trajectories, you can 
`rostopic echo /my_gen3/gen3_joint_trajectory_controller/follow_joint_trajectory/goal` 
when you send a trajectory by MoveIt (you can do this by RViz, or by launching the [MoveIt Python example](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_examples/launch/moveit_example.launch)).

MoveIt was configured to send trajectories that the arm would accept, but for trajectories generated with another framework, you will need to follow some guidelines described in [this documentation](https://github.com/Kinovarobotics/matlab_kortex/blob/master/simplified_api/documentation/precomputed_joint_trajectories.md) in our other repository (matlab_kortex). Basically, the trajectory needs to be computed with a timestep of 1ms, and the velocity and acceleration commands of each joints cannot be higher than their limits, or the arm will reject the trajectory.

Don't hesitate if you have other questions,
Best,
Alex