Kinovarobotics/ros_kortex #29: Trajectory control

akihikoy, 2019-06-11:
Although this was mentioned in https://github.com/Kinovarobotics/ros_kortex/issues/27, I would like to keep this issue opened.

At this moment, we cannot control trajectory of Gen3, which is a lack of fundamental feature of robots.  Everyone should be able to know this fact before buying the robot.  So, I would like to keep this open.

simonferronforget, 2019-10-17:
Hello,
The [**FollowJointTrajectory** ](https://github.com/Kinovarobotics/ros_kortex/blob/v2.0.0/kortex_driver/src/non-generated/pre_computed_joint_trajectory_action_server.cpp) server in ros_kortex was introduced in last release. Did you have the chance to try it ? Does it fit your needs ?

Regards,
Simon

akihikoy, 2019-11-26:
@simonferronforget 

As mentioned in https://github.com/Kinovarobotics/kortex/issues/16 by @alexvannobel :

- The Base service now supports the [PlayPreComputedJointTrajectory function](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/summary_pages/Base.md).
- ROS version can be found in the [ROS FollowJointTrajectory server of ros_kortex](https://github.com/Kinovarobotics/ros_kortex/blob/v2.0.0/kortex_driver/src/non-generated/pre_computed_joint_trajectory_action_server.cpp).
- The note is that there are limitations to use it as described [here](https://github.com/Kinovarobotics/matlab_kortex/blob/master/simplified_api/documentation/precomputed_joint_trajectories.md#hard-limits-and-conditions-to-respect).

I explored this action server from ROS.

**My input is a set of via points of a trajectory** (set of (joint angles, time from start) pairs) as usual ROS follow_joint_trajectory.
We use the `/my_gen3/gen3_joint_trajectory_controller/follow_joint_trajectory` action server to follow a joint angular trajectory.  However, the trajectory should have 1msec timesteps intervals between all trajectory points.  In order to satisfy this limitation, I used [cubic Hermite splines](https://en.wikipedia.org/wiki/Cubic_Hermite_spline) to generate 1 msec interval  trajectory from my discrete via points.

The test code can be found: https://github.com/akihikoy/ay_test/blob/master/ros/py_ros/gen3/follow_q_traj3.py
The video: https://youtu.be/CUByCHOaKcE

It is almost working.
**However there is still an issue**.
When the goal pose (the final target) is the same as the current pose (or within the `goal_tolerance`), it cancels the trajectory with the message: "We already reached the goal position : nothing to do."
This should be a bug.
We cannot make cyclic trajectories.

alexvannobel, 2019-11-26:
Hi @akihikoy ,

A couple of MoveIt bugs were flagged to us (bugs with the action server in fact) and we'll look into them very shortly. We will probably push the fixes to the kinetic branch in the coming weeks. 
Meanwhile, you could comment out [these lines](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/src/non-generated/pre_computed_joint_trajectory_action_server.cpp#L83-L88) in the action server, and recompile it. This will remove the check on the "goal already reached".

I'll update this issue when we have a definitive fix for that.
Hope this helps,
Alex

akihikoy, 2019-11-27:
Hi @alexvannobel ,

Thanks.  By commenting [those lines](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/src/non-generated/pre_computed_joint_trajectory_action_server.cpp#L83-L88), I was able to create cyclic motions.

Let's keep this issue open until it is officially fixed.
Akihiko

akihikoy, 2020-02-19:
Hello @alexvannobel ,

I would like to know whether the [hard limits](https://github.com/Kinovarobotics/matlab_kortex/blob/master/simplified_api/documentation/precomputed_joint_trajectories.md#hard-limits-and-conditions-to-respect) of ``/my_gen3/gen3_joint_trajectory_controller/follow_joint_trajectory`` are removed.

Especially the limitation of time increments (_Time increments must be 0.001 seconds between trajectory points_) is bothering me.

Thanks,
--Akihiko

akihikoy, 2020-02-19:
I also noticed that there is an interval (1.0-2.0 sec) between the trajectory command request and the actual motion start.

More clearly, after the command

    client.send_goal(goal)

is executed (`client`: `SimpleActionClient`, `goal`:`FollowJointTrajectoryGoal`), it takes 1.0-2.0 sec to start the motion.

Perhaps the trajectory controller takes time to process the large amount of data (as the trajectory points are generated at each 0.001 sec).  But anyway, this is not good for many of robot applications.

Any solutions, or did you already solve this issue?

Thanks,
--Akihiko

elevenjiang1, 2021-01-28:
> I also noticed that there is an interval (1.0-2.0 sec) between the trajectory command request and the actual motion start.
>
> More clearly, after the command
>
> ```
> client.send_goal(goal)
> ```
>
> is [...]

Hello,I reference your ay_test code of gen3 and can control the robot by /my_gen3/gen3_joint_trajectory_controller/follow_joint_trajector
However,it seems that the robot can only move very very slow.The resaon i guess may be too much position for robots and the communication between robot and computer loss.
```
Its error is:
Type: TRAJECTORY_ERROR_TYPE_INVALID_ACCELERATION
Identifier:TRAJECTORY_ERROR_TYPE_INVALID_ACCELERATION
Actuator:4
Erroneous value is 86.4247 but minimun permitted is 0 and maximum permitted is 51.5662
```

Any solution for this?Thank you!

elevenjiang1, 2021-02-05:
I find that use FollowJointTrajectory to control robot may occur the robot block.
I just simply let the robot move repeatedly ,after 10min,the robot can not move,and the webapp can not connect.

alexvannobel, 2021-06-04:
Hi to all,

The latest release (v2.3.0) adds Waypoints API to control the Kinova Gen3 and Gen3 lite.

The `FollowJointTrajectory` action server was modified to use this API and support via points that have arbitrary durations between themselves.

I'll close this issue since the initial request was fulfilled.

Best,
Alex

closed 2021-06-04 by alexvannobel