Kinovarobotics/ros_kortex #33: Error while launching the kortex driver

RaduCorcodel, 2019-08-13:
As the title says, after launching the kortex driver (rel 2.0.0) with 

`roslaunch kortex_driver kortex_driver.launch ip_address:=1xx.yyy.zzz.tt gripper:="robotiq_2f_85"`

the output returns the following error:

`[ERROR] [1565733224.731845080]: Exception while loading planning adapter plugin 'industrial_trajectory_filters/UniformSampleFilter': According to the loaded plugin descriptions the class industrial_trajectory_filters/UniformSampleFilter with base class type planning_request_adapter::PlanningRequestAdapter does not exist. Declared types are  default_planner_request_adapters/AddIterativeSplineParameterization default_planner_request_adapters/AddTimeParameterization default_planner_request_adapters/Empty default_planner_request_adapters/FixStartStateBounds default_planner_request_adapters/FixStartStateCollision default_planner_request_adapters/FixStartStatePathConstraints default_planner_request_adapters/FixWorkspaceBounds`

any advice?

sea-bass, 2019-08-14:
I got around this by setting **start_moveit:=false** ... from searching around, it seems like a MoveIt! version issue?

alexvannobel, 2019-08-14:
Hi,

I can reproduce your error if I uninstall `ros-kinetic-industrial-trajectory-filters`.
Can you go to the root of your catkin workspace and run ```rosdep install --from-paths src --ignore-src```? This will install the missing dependencies for ros_kortex. Some packages, as this ros-industrial package are now used for MoveIt support. 

I'll update the release notes instructions so users also install the new dependencies.

Does this solve your problem?

sea-bass, 2019-08-14:
@alexvannobel -- confirming that after installing that missing package, the driver starts correctly with **start_moveit:=true**

closed 2019-08-14 by alexvannobel

RaduCorcodel, 2019-08-14:
After installing the missing dependency the driver starts error-free.

huiwenzhang, 2020-04-03:
@alexvannobel  Hi, I encounter the same error when I tried to start the kinova gazebo tool by running 'roslaunch kortex_gazebo spawn_kortex_robot.lauch'. I followed the instructions pasted above and confirmed that the ros-kinetic-industrial-core and ros-kinetic-industrial-trajectory-filters have be successfully installed, the version is 0.7.1. But the error still exists, do you know what's going  wrong there?

polthergeist, 2020-04-06:
Hi @huiwenzhang ,
Is it the exact same error as posted by @RaduCorcodel ?

alexvannobel, 2020-04-21:
Hi @huiwenzhang ,

If you have exactly the same error as stated by the users above, most probably this is the same dependency problem. Can you make sure the dependencies are installed by going to the root of the cakin workspace and running:

`rosdep install --from-paths src --ignore-src -y`

If the error persists after cleaning and rebuilding the workspace, can you upload your build ouput (terminal output when running `catkin_make`) on this thread?

Best,

Alex