Kinovarobotics/ros_kortex #34: Compilation error with multiple versions of protobuf

juntuck, 2019-08-16:
Greetings,

I'm trying to install the latest ros_kortex packages, but I'm facing some issues.
When I catkin_make my workspace, I get a multitude of errors with the two repeating templates.

### **Error Template 1:**

_In file included from /usr/include/ignition/msgs1/ignition/msgs/MessageTypes.hh:164:0,
                 from /usr/include/ignition/msgs1/ignition/msgs/Utility.hh:29,
                 from /usr/include/ignition/msgs1/ignition/msgs.hh:25,
                 from /usr/include/ignition/transport4/ignition/transport/Node.hh:32,
                 from /usr/include/gazebo-9/gazebo/physics/Entity.hh:25,
                 from /usr/include/gazebo-9/gazebo/physics/Model.hh:30,
                 from /usr/include/gazebo-9/gazebo/physics/Actor.hh:26,
                 from /usr/include/gazebo-9/gazebo/physics/physics.hh:2,
                 from /home/judejtng/cronos_ws/src/ros_kortex/third_party/gazebo-pkgs/gazebo_version_helpers/src/GazeboVersionHelpers.cpp:2:
/usr/include/ignition/msgs1/ignition/msgs/visual_v.pb.h:12:2: error: #error **This file was generated by a newer version of protoc which is**
 #error This file was generated by a newer version of protoc which is
  ^~~~~
/usr/include/ignition/msgs1/ignition/msgs/visual_v.pb.h:13:2: error: #error **incompatible with your Protocol Buffer headers. Please update**
 #error incompatible with your Protocol Buffer headers.  Please update
  ^~~~~
/usr/include/ignition/msgs1/ignition/msgs/visual_v.pb.h:14:2: error: #error **your headers.**
 #error your headers._

### **Error Template 2:**

In file included from /usr/include/google/protobuf/stubs/macros.h:34:0,
                 from /usr/include/google/protobuf/stubs/logging.h:34,
                 from /usr/include/google/protobuf/arena.h:55,
                 from /usr/include/ignition/msgs1/ignition/msgs/color.pb.h:22,
                 from /usr/include/gazebo-9/gazebo/msgs/msgs.hh:32,
                 from /usr/include/gazebo-9/gazebo/gazebo_core.hh:20,
                 from /usr/include/gazebo-9/gazebo/gazebo.hh:20,
                 from /home/judejtng/cronos_ws/src/ros_kortex/third_party/roboticsgroup_gazebo_plugins/include/roboticsgroup_gazebo_plugins/mimic_joint_plugin.h:37,
                 from /home/judejtng/cronos_ws/src/ros_kortex/third_party/roboticsgroup_gazebo_plugins/src/mimic_joint_plugin.cpp:23:
**/usr/include/google/protobuf/stubs/port.h**:145:21: error: **redefinition** of ‘const uint32 google::protobuf::kuint32max’
 static const uint32 kuint32max = 0xFFFFFFFFu;
                     ^~~~~~~~~~
In file included from /usr/include/ignition/msgs1/ignition/msgs/color.pb.h:9:0,
                 from /usr/include/gazebo-9/gazebo/msgs/msgs.hh:32,
                 from /usr/include/gazebo-9/gazebo/gazebo_core.hh:20,
                 from /usr/include/gazebo-9/gazebo/gazebo.hh:20,
                 from /home/judejtng/cronos_ws/src/ros_kortex/third_party/roboticsgroup_gazebo_plugins/include/roboticsgroup_gazebo_plugins/mimic_joint_plugin.h:37,
                 from /home/judejtng/cronos_ws/src/ros_kortex/third_party/roboticsgroup_gazebo_plugins/src/mimic_joint_plugin.cpp:23:
**/home/judejtng/cronos_ws/src/ros_kortex/third_party/roboticsgroup_gazebo_plugins/../../kortex_gazebo/include/google/protobuf/stubs/common.h**:201:21: note: ‘const uint32 google::protobuf::kuint32max’ previously defined here
 static const uint32 kuint32max = 0xFFFFFFFFu;

### **Question:**

What could be the source of the errors? It seems to me that two versions of protobuf are visible to the compiler - one included in kortex_api/include/google/protobuf, the other in /usr/include/google/protobuf - and the compiler is confused.

$ protoc --version 

tells me I have version 3.0.0 installed. I think it was automatically installed when I installed Gazebo a while back.

Has anyone experienced this issue and/or can point me in the right direction to solving it? Thanks! :)

Best regards,
Jude

alexvannobel, 2019-08-16:
Hi @juntuck ,

You seem to be on Gazebo 9. Are you on ROS Melodic? The package was tested on ROS Kinetic and Gazebo 7, which use Protobuf 2.6 I believe. This is why I added the include and lib files for this specific version of Protobuf (so if people have another Protobuf version installed on their system, it still links correctly).
We plan on working on Melodic support soon, but in the meantime can you test if removing the kortex_gazebo/include/google and kortex_gazebo/lib folders can solve your error? You may have to modify the third_party plugins CMakeLists.txt so they stop looking for these folders. 

Let me know how this goes,
Best,
Alex

juntuck, 2019-08-16:
Dear Alex,

I am indeed working with ROS Melodic. I removed both folders as you suggested and it worked! :) Did not have to modify third_party plugins CMakeLists.txt.

Many thanks for the prompt advice!

Best regards,
Jude

closed 2019-08-16 by alexvannobel