Kinovarobotics/ros_kortex #36: kortex_driver launching error

kracon7, 2019-08-20:
Hi, 
I'm using release 2.0.0 on ubuntu 16.04 and ros-kinetic.
I'm connecting to Gen3 with ether-net cable and set up the ip address and netmask according to the [quick start guide](https://www.kinovarobotics.com/sites/default/files/QS-002_KINOVA_Gen3_Ultra_lightweight_robot-%20Quick_Start_Guide_WEB_EN_R01%20%281%29.pdf).
However when I tried to do ``` roslaunch kortex_driver kortex_driver.launch start_rviz:=false start_moveit:=false ip_address:=192.168.1.11```  I got an error message as follows:

```sh
jc@jc-Precision-5510:~$ roslaunch kortex_driver kortex_driver.launch start_rviz:=false start_moveit:=false ip_address:=192.168.1.11
... logging to /home/jc/.ros/log/7725a7b6-c37e-11e9-8537-a434d9e04d1d/roslaunch-jc-Precision-5510-14874.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jc-Precision-5510:42769/

SUMMARY
========

PARAMETERS
 * /my_gen3/joint_state_publisher/source_list: ['base_feedback/j...
 * /my_gen3/my_gen3_driver/api_connection_inactivity_timeout_ms: 20000
 * /my_gen3/my_gen3_driver/api_rpc_timeout_ms: 2000
 * /my_gen3/my_gen3_driver/api_session_inactivity_timeout_ms: 35000
 * /my_gen3/my_gen3_driver/arm: gen3
 * /my_gen3/my_gen3_driver/cyclic_data_publish_rate: 100
 * /my_gen3/my_gen3_driver/default_goal_time_tolerance: 0.5
 * /my_gen3/my_gen3_driver/default_goal_tolerance: 0.5
 * /my_gen3/my_gen3_driver/gripper: 
 * /my_gen3/my_gen3_driver/ip_address: 192.168.1.11
 * /my_gen3/my_gen3_driver/joint_names: ['joint_1', 'join...
 * /my_gen3/robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /my_gen3/
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    my_gen3_driver (kortex_driver/kortex_arm_driver)
    robot_state_publisher (robot_state_publisher/state_publisher)

auto-starting new master
process[master]: started with pid [14887]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7725a7b6-c37e-11e9-8537-a434d9e04d1d
process[rosout-1]: started with pid [14900]
started core service [/rosout]
process[my_gen3/my_gen3_driver-2]: started with pid [14903]
process[my_gen3/joint_state_publisher-3]: started with pid [14909]
process[my_gen3/robot_state_publisher-4]: started with pid [14919]
There was a problem connecting to the server. Aborting
[ERROR] [1566328355.569615805]: The node could not connect to the arm. Did you specify the right IP address and is the arm powered on?
terminate called after throwing an instance of 'std::runtime_error'
  what():  not connected !!!
[my_gen3/my_gen3_driver-2] process has died [pid 14903, exit code -6, cmd /home/jc/catkin_ws/devel/lib/kortex_driver/kortex_arm_driver __name:=my_gen3_driver __log:=/home/jc/.ros/log/7725a7b6-c37e-11e9-8537-a434d9e04d1d/my_gen3-my_gen3_driver-2.log].
log file: /home/jc/.ros/log/7725a7b6-c37e-11e9-8537-a434d9e04d1d/my_gen3-my_gen3_driver-2*.log
```

I tried the default ip address but the process just hang for a little longer and die with same error message show above. Any thoughts on this one? Thanks in advance.

RaduCorcodel, 2019-08-20:
You said you setup an IP address for the robot (possibly a different one from default 192.168.1.10?). What is the IP address the arm has now? That one has to match the argument  `ip_address:=` in the roslaunch. Can you connect to the robot through the Chrome WebApp?

P.S. the robot's default IP address is 192.168.1.10. The address 192.168.1.11 is an example for the host computer (but I guess it can be anything in the 192.168.1.xxx family)

RaduCorcodel, 2019-08-20:
Looking at you launch command, I think I spotted the problem. 
Assuming you didn't change the default IP address, try this:
`roslaunch kortex_driver kortex_driver.launch start_rviz:=false start_moveit:=false ip_address:=192.168.1.10`

kracon7, 2019-08-20:
Hi @RaduCorcodel ,
I set the ip address as "192.168.1.11"
The Chrome WebApp can connet without an error using ip address "192.168.1.10"
I tried the command you gave me where ip address is 192.168.1.10, the error remains.

RaduCorcodel, 2019-08-20:
So you can connect to the robot using the WebApp at IP 192.168.1.10? If yes, that's the IP of the robot.

kracon7, 2019-08-20:
@RaduCorcodel 
Yes I could use the WebApp.
But if I try roslaunch it fails on that ip...
The error message now is:
```sh
jc@jc-Precision-5510:~$roslaunch kortex_driver kortex_driver.launch start_rviz:=false start_moveit:=false
... logging to /home/jc/.ros/log/183da958-c390-11e9-8537-a434d9e04d1d/roslaunch-jc-Precision-5510-20002.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jc-Precision-5510:46075/

SUMMARY
========

PARAMETERS
 * /my_gen3/joint_state_publisher/source_list: ['base_feedback/j...
 * /my_gen3/my_gen3_driver/api_connection_inactivity_timeout_ms: 20000
 * /my_gen3/my_gen3_driver/api_rpc_timeout_ms: 2000
 * /my_gen3/my_gen3_driver/api_session_inactivity_timeout_ms: 35000
 * /my_gen3/my_gen3_driver/arm: gen3
 * /my_gen3/my_gen3_driver/cyclic_data_publish_rate: 100
 * /my_gen3/my_gen3_driver/default_goal_time_tolerance: 0.5
 * /my_gen3/my_gen3_driver/default_goal_tolerance: 0.5
 * /my_gen3/my_gen3_driver/gripper: 
 * /my_gen3/my_gen3_driver/ip_address: 192.168.1.10
 * /my_gen3/my_gen3_driver/joint_names: ['joint_1', 'join...
 * /my_gen3/robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /my_gen3/
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    my_gen3_driver (kortex_driver/kortex_arm_driver)
    robot_state_publisher (robot_state_publisher/state_publisher)

auto-starting new master
process[master]: started with pid [20015]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 183da958-c390-11e9-8537-a434d9e04d1d
process[rosout-1]: started with pid [20028]
started core service [/rosout]
process[my_gen3/my_gen3_driver-2]: started with pid [20031]
process[my_gen3/joint_state_publisher-3]: started with pid [20039]
process[my_gen3/robot_state_publisher-4]: started with pid [20047]
There was a problem connecting to the server. Aborting
[ERROR] [1566336058.247137988]: The node could not connect to the arm. Did you specify the right IP address and is the arm powered on?
terminate called after throwing an instance of 'std::runtime_error'
  what():  not connected !!!
[my_gen3/my_gen3_driver-2] process has died [pid 20031, exit code -6, cmd /home/jc/catkin_ws/devel/lib/kortex_driver/kortex_arm_driver __name:=my_gen3_driver __log:=/home/jc/.ros/log/183da958-c390-11e9-8537-a434d9e04d1d/my_gen3-my_gen3_driver-2.log].
log file: /home/jc/.ros/log/183da958-c390-11e9-8537-a434d9e04d1d/my_gen3-my_gen3_driver-2*.log

```

alexvannobel, 2019-08-20:
Hi @kracon7 ,

This is a weird behaviour. Normally if you can ping the arm and connect to the Webapp, you can also start the ROS driver with this IP address.
The only reason I would see the driver not being able to connect and the WebApp being able to connect would be a mismatch between the Kortex API used by ROS and the robot's API. This would happen if the firmware update failed (or was not done) or if the ROS packages are not up-to-date (especially the API).
To make sure the update to 2.0.0 worked : 
- Verify the firmware in the robot was updated to 2.0.0-60 for all the sub-devices (in the upgrade page)

To make sure the ROS packages are up-to-date:
- Delete the kortex_api/include and kortex_api/lib folders
- Pull the kinetic-devel branch to the latest commit
- Delete the build and devel folders in the root of your catkin workspace
- `catkin_make` again

Hope this helps,
Best,
Alex

kracon7, 2019-08-21:
@alexvannobel 
It is due to the mismatch of firmware. Now connection is successful. Thanks!

closed 2019-08-21 by kracon7