Kinovarobotics/ros_kortex #39: running ros node error

leejiawei, 2019-09-04:
Hi:

I got errors:

[ERROR] [1567627554.028590077]: The gripper model specified in the launch file doesn't match the detected arm's gripper model, shutting down the node...
terminate called after throwing an instance of 'std::runtime_error*'
[my_gen3/my_gen3_driver-1] process has died [pid 26321, exit code -6, cmd /home/airlab/catkin_ws/devel/lib/kortex_driver/kortex_arm_driver __name:=my_gen3_driver __log:=/home/airlab/.ros/log/481ae8aa-cf4f-11e9-97c4-94c691adb6c2/my_gen3-my_gen3_driver-1.log].
log file: /home/airlab/.ros/log/481ae8aa-cf4f-11e9-97c4-94c691adb6c2/my_gen3-my_gen3_driver-1*.log

when I run command  roslaunch kortex_driver kortex_driver.launch

alexvannobel, 2019-09-04:
Hi @leejiawei ,

The kortex_driver checks the arm's product configuration and compares it to the input of the launch file to make sure they match.
The default gripper in the launch file is no gripper.
There are two possibilities here with the command line you use to launch the driver : 
1.  **If your arm doesn't have a Robotiq gripper connected**, then the Product Configuration for the arm is wrong and you can change it in the WebApp. If you go to Robot ---> Configuration ---> Product, you can change the End Effector Type field to Not Installed. You can reboot the arm and relaunch the driver with the same command line you used and it should work.
2. **If your arm has a Robotiq gripper connected**, then you should give it the gripper name as an  argument in the launch file:
``` roslaunch kortex_driver kortex_driver gripper:=robotiq_2f_85``` 
The driver should launch correctly then.

Hope this helps, 
Best,
Alex

leejiawei, 2019-09-04:
Thanks for the help

I will try it soon

leejiawei, 2019-09-04:
It works 

Thanks for the help

closed 2019-09-04 by alexvannobel