Kinovarobotics/ros_kortex #40: Compilation error

abhisek-roy, 2019-09-14:
Hi,
I was trying to compile the ros_kortex package, but running into following error:

[ 21%] Building CXX object ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/src/generated/base_services.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/build.make:230: recipe for target 'ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/src/generated/base_services.cpp.o' failed
make[2]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/src/generated/base_services.cpp.o] Error 4
CMakeFiles/Makefile2:5378: recipe for target 'ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/all' failed
make[1]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1 -l1" failed

I have setup ROS Melodic and I am running Ubuntu 18.04 in VMware 15.

alexvannobel, 2019-09-16:
Hi @abhisek-roy ,
This is a weird error, the error source does not really show in the log you sent. 
I have a couple questions:
- Was there other error messages before this snippet in the full log?
- What is your version of GCC? You can run ```gcc --version```
- Did the Kortex API download correctly? What is the content of kortex_api/include and kortex_api/lib?
- Do you have the correct write permissions on the directory in the VM?

Best, 
Alex

abhisek-roy, 2019-09-16:
Hi @alexvannobel ,

Thanks for your response. Please find my responses below.

- Was there other error messages before this snippet in the full log?
             I don't think I have noticed any warnings or errors prior to the quotes error message. I have added a text file with the output of the commands.

- What is your version of GCC? You can run gcc --version
             gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0

- Did the Kortex API download correctly? What is the content of kortex_api/include and kortex_api/lib?
             I redownloaded it and it seems it has been downloaded fine. The git command executed fine. Please check the attached txt file.

- Do you have the correct write permissions on the directory in the VM?
             I think I have the proper write permissions as well. I am working in the home folder and I have 6 GB available space.

[terminal.txt](https://github.com/Kinovarobotics/ros_kortex/files/3617336/terminal.txt)

abhisek-roy, 2019-09-23:
The issue was happening due to out of memory. The issue is captured here and the solution works.

https://stackoverflow.com/questions/30887143/make-j-8-g-internal-compiler-error-killed-program-cc1plus

Closing the issue.

closed 2019-09-23 by abhisek-roy