Kinovarobotics/ros_kortex #44: Sending a list of waypoints to the arm

RaduCorcodel, 2019-09-25:
I would like to send a set of waypoints for the robot arm to execute (not in Gazebo, but the real hardware). I see there is an example showing how to send one goal, but how about if I have a number of via points that I compute separately? They are usually 7D points in the joint space. On other robots, I can send them over to the Joint Trajectory Controller. Is there a way to achieve this on our Kinova Gen3 arm?

RaduCorcodel, 2019-10-05:
Reviving this topic since it's important I believe. 
Can anyone give me a hint on how to use the PreComputedJointTrajectoryActionServer? A small example would go a long way.

alexvannobel, 2019-10-08:
Hi @RaduCorcodel ,
Sorry for delay.
The Gen3 arm does not support at the moment sending Waypoints to it, it's going to be in a next release. I'll mark this thread as "not implemented yet".
The PreComputedJointTrajectory is a way to achieve the same effect, but you will need to supply the server with joint angles for every millisecond of the trajectory, that the arm will validate and then execute if it passes the validation (max speed, max acceleration). 
So, to send a trajectory to the arm, you will need to supply one goal with 7D points every one millisecond (ex. a 3.2s trajectory will need to contain 3200 trajectory points, the first point starting at t=0.000 and the second =0.001, etc.). We made some documentation about the PreComputed Joint Trajectories in our Matlab API repo, [you can have a look at it here](https://github.com/Kinovarobotics/matlab_kortex/blob/master/simplified_api/documentation/precomputed_joint_trajectories.md). 
I also invite you to take a look at [the source code](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/src/non-generated/pre_computed_joint_trajectory_action_server.cpp) of the Action Server. You can add prints in the code and observe the output when you send trajectories with MoveIt to see what kind of trajectories MoveIt sends, and the way the goal you send should look.

Best,
Alex

RaduCorcodel, 2019-10-08:
Thanks for the response. I think this should get more attention from the developing team since it's a basic functionality of any robot. We got this arm quite some time ago and it starts to be a big problem that we can't implement basic control algorithms.
Can we please have at least a release schedule so we know how to allocate resources? We can't have this much uncertainty about we can do with this robot.

alexvannobel, 2019-10-09:
Hi @RaduCorcodel ,

We are actually working on the waypoints implementation for the next release, normally due at the end of this year.
However in the meantime, there are a couple options already available to you to implement "basic control algorithms":
- You can use the SendJointSpeedsCommand service (or RPC call from the C++ or Python API directly) to control the arm in angular velocity.
- You can use the SendTwistCommand service (or RPC call from the C++ or Python API directly) to control the arm in Cartesian velocity.
- You can interpolate your via-points into a 1ms-interval trajectory and use the action server described above.
- If you use the C++ API standalone (not supported in ROS yet), you can use low-level real-time calls at 1kHz to BaseCyclic->Refresh to control every actuator's position in real-time. This requires a real-time capable OS, though.

I'll leave this issue open and make sure to update it once the feature becomes available, but if you want more solutions in the meantime, or customized solutions to control you Gen3 arm, I invite you to contact support@kinova.ca. 

Best regards, 
Alex

RaduCorcodel, 2019-10-09:
I started looking exactly at these options since I opened the issue and should have something working pretty soon. So far it's rejecting the interpolation points either due to velocity constraints or the 1kHz rate but I'll keep on it. Thank you for your reply, I really appreciate pointing me in the right direction. Looking forward for the next release.

DiegoRomeres, 2020-11-24:
Is there any update on this topic? is it possible now to give the robot a full trajectory in either joint or cartesian space and have the robot execute it (without having to specify the 1kHz via points)?

alexvannobel, 2021-06-04:
Hi to all,

It is now possible with release 2.3.0 to send a full trajectory to the arm either in Cartesian or Joint space.

The `FollowJointTrajectory` action server was modified to use the new Waypoints API in the release and a Cartesian trajectory action server was added, with corresponding [examples](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_examples/src/full_arm/example_waypoint_action_client.cpp).

I'll close this issue since this feature request has been fulfilled.

Best,
Alex

closed 2021-06-04 by alexvannobel