Kinovarobotics/ros_kortex #45: Vision topic activation service doens't seem to work

RaduCorcodel, 2019-10-09:
I'm trying to use the vision module (I'm more interested in the depth point cloud). As I understood, I need to activate the publisher. So I wrote this node:

```
void notification_callback(const kortex_driver::VisionNotification::ConstPtr& notif) {
    ROS_INFO("Got data");
}

void activate_vision_topic(ros::NodeHandle n, std::string robot_name) {
    // See if we have the vision module connected
    ros::ServiceClient service_client_read_all_devices = n.serviceClient<kortex_driver::ReadAllDevices>("/" + robot_name + "/device_manager/read_all_devices");
    kortex_driver::ReadAllDevices service_read_all_devices;

    if (service_client_read_all_devices.call(service_read_all_devices)) {
        auto output = service_read_all_devices.response.output;
        for (int i=0; i<output.device_handle.size(); i++) {
            if (output.device_handle[i].device_type == kortex_driver::DeviceTypes::VISION) {
                ROS_INFO_STREAM("VISION MODULE >> DevID:" << output.device_handle[i].device_identifier <<
                                                " DevType: " << output.device_handle[i].device_type <<
                                                " DevOrder: " << output.device_handle[i].order);
                break;
            }
        }
    }

    //Activate the vision topic
    std::string VISION_TOPIC_ACTIVATION_NAME = "/" + robot_name + "/vision_config/activate_publishing_of_vision_topic";
    ros::ServiceClient service_client_activate_notif = n.serviceClient<kortex_driver::OnNotificationVisionTopic>(VISION_TOPIC_ACTIVATION_NAME);
    kortex_driver::OnNotificationVisionTopic service_activate_notif;
    service_activate_notif.request.input.rate_m_sec = 33;
    service_activate_notif.request.input.threshold_value = 0.01f; // Don't know what this does
    service_activate_notif.request.input.type = kortex_driver::Sensor::SENSOR_DEPTH;
   
    if (service_client_activate_notif.call(service_activate_notif)) {
        ROS_INFO_STREAM("Publisher activated");
    }
    else { ROS_ERROR_STREAM("Cannot activate"); }
    ros::Duration(1.0).sleep();
}

int main(int argc, char **argv) {
  ros::init(argc, argv, "activate_vision");
  ros::NodeHandle n;
  std::string robot_name = "my_gen3";
  ros::Subscriber sub = n.subscribe("/" + robot_name  + "/vision_topic", 1000, notification_callback);

  activate_vision_topic(n, robot_name);
  ros::spin();
  return 0;
}
```
About half the time I get the error:
```
[ INFO] [1570573329.615870357]: Kortex exception
[ INFO] [1570573329.615934206]: KINOVA exception error code: 1
[ INFO] [1570573329.615987718]: KINOVA exception error sub code: 52
[ INFO] [1570573329.616049492]: KINOVA exception description: Server protocol error, Error sub type=INVALID_DEVICE => <srv: 5, fct: 8, msgType: 3> description: The device id 255 is invalid and doesn't correspond to a registered device.
```
The other half, there are no errors but there is nothing being published under the `/my_gen3/vision_topic` topic regardless.

Am I missing something?

alexvannobel, 2019-10-09:
Hi @RaduCorcodel ,

The [ros_kortex_vision repository](https://github.com/Kinovarobotics/ros_kortex_vision) is the repository you need to access the color and depth streams of the Gen3 camera. I admit it is not clear right now how to access the video streams, and I will add something in the ros_kortex readme file to link to this other repository. We will eventually merge this other repository into ros_kortex.
The "vision topic" is the arm's notification topic for vision-configuration related events, but I don't think the vision module sends notifications at the moment. 
I'll close this issue, but if you encounter any problem using ros_kortex_vision, feel free to open an issue in this repo.
I'll also create a general issue about "how to read the video streams" and pin it in ros_kortex so people can see it.

Thanks for the feedback,
Best regards, 
Alex

closed 2019-10-09 by alexvannobel

RaduCorcodel, 2019-10-09:
Great! Thank you, I will try the vision repo and report back.

alexvannobel, 2019-10-09:
Good!
I created and pinned #46 .