Kinovarobotics/ros_kortex #47: Velocity control in the simulation

juntuck, 2019-10-18:
Hello there! :)

I would like to control the Gazebo-simulated arm by sending Twist msgs to the '/my_gen3/base/send_twist_command'.

For the real robot, kortex_driver.launch starts this service. However, this launch file seems to always require a real robot running.

Is there a way to get kortex_driver working with the simulation? Or did I perhaps miss an equivalent service or topic in the simulation?

Best regards,
Jude

alexvannobel, 2019-10-21:
Hi @juntuck ,

The kortex_driver package won't work in simulation. You will have to use the [kortex_gazebo](https://github.com/Kinovarobotics/ros_kortex/tree/kinetic-devel/kortex_gazebo) package to simulate the Gen3 arm. I invite you to read the README file for the package, as it is still experimental. 

Currently (v2.0.0), we do not support the Kortex API services in Gazebo. The only simulation interface is via ROS Control position controllers for every actuator (JointPositionController's) or for the whole arm (JointTrajectoryController). 
Hence, you won't be able to use the '/my_gen3/base/send_twist_command' service to send a Cartesian speed to the robot in Gazebo. 

For now, I think your best option will be to use KDL to calculate the inverse kinematics for the robot, and send increments of angular positions to every actuator via the ROS Control interface to achieve the Cartesian speeds you want. 

Best,
Alex

juntuck, 2019-11-14:
Dear @alexvannobel ,

Thank you for your input on the matter. A short update:

I made do by sending Twist msgs directly to the arm, without visualizing it in RVIZ.
My aim was two-fold:
1) to generate a Cartesian trajectory of the end-effector to a goal pose that is a line - The SendTwistCommand service works perfectly for this; I computed delta_x, _y and _z between goal and current pose, v_x, _y and _z with my desired duration, and sent those values in the Twist msg. I send a second Twist msg with all 0's after said duration.
2) to visualize movement of the arm joints before executing the trajectory, in case they do anything odd - I suppose this is unnecessary for now, as the arm accounts for self-collision and the elbow doesn't seem to drastically flip around around as I feared. 

Nevertheless, it became apparent to me that MoveIt has a feature for generating straight line Cartesian paths. The generated trajectory can be viewed with /move_group/display_planned_path topic in RVIZ, but somehow the Gen3 arm refuses to execute it. Any idea why?

Best regards,
Jude

alexvannobel, 2019-11-14:
Hi @juntuck ,

A couple points before addressing your MoveIt problem:
- If you want the arm to reach a Cartesian goal within a specific duration, you can use the `.../base/play_cartesian_trajectory` service and specify a constraint duration in the Request.
- You may experience a better "stop" behaviour if you stop the arm with the `.../base/stop` service instead of sending a null twist (the arm will quickly decelerate and stop, instead of jerking a bit and stopping).

As for your MoveIt problem, can you post the console output of the kortex_driver when the error happens and also the output of your MoveIt client node? Maybe the arm rejects the trajectory because of an invalid speed, or acceleration, but we'll see it in the kortex_driver output. 

Best,
Alex

juntuck, 2019-11-28:
Dear @alexvannobel ,

Thanks for the two suggestions! I took your advice and they work perfectly. :)

For the MoveIt problem, I plan to tackle it again this week. I suspect it's because MoveIt does not generate enough points to satisfy the Kortex controller. Nevertheless, I will post my updates as soon as I get back into the thick of things.

Best,
Jude

alexvannobel, 2020-01-28:
Hi @juntuck  ,

I'll close this issue for inactivity, but feel free to re-open it if it persists or to open a new issue.

Best regards,
Alex

closed 2020-01-28 by alexvannobel