Kinovarobotics/ros_kortex #48: catkin_make install error

mlauret, 2019-10-18:
Hello !

I got an error trying to use the install verb for catkin_make : 

`catkin_make install`

I got an error for some packages : 

**For kortex_control :** 
```
CMake Error at ros_kortex/kortex_control/cmake_install.cmake:51 (file):
  file INSTALL cannot find
  "/home/roso/Workspaces/kortex/src/ros_kortex/kortex_control/config".
Call Stack (most recent call first):
  cmake_install.cmake:130 (include)

```
In the CmakeLists, the line : 

`foreach(dir config launch meshes urdf)`

could be : 

`foreach(dir arms grippers)`

(at least, I have no error with the new line and the package install correctly)

**for kortex_description :** 

CMake Error at ros_kortex/kortex_description/cmake_install.cmake:51 (file):
  file INSTALL cannot find
  "/home/roso/Workspaces/kortex/src/ros_kortex/kortex_description/config".
Call Stack (most recent call first):
  cmake_install.cmake:131 (include)

the line : 

`foreach(dir config launch meshes urdf)`

could be : 

`foreach(dir arms grippers robots)`

**for kortex_gazebo :** 

```
CMake Error at ros_kortex/kortex_gazebo/cmake_install.cmake:51 (file):
  file INSTALL cannot find
  "/home/roso/Workspaces/kortex/src/ros_kortex/kortex_gazebo/config".
Call Stack (most recent call first):
  cmake_install.cmake:132 (include)
```

the line : 

`foreach(dir config launch meshes urdf)`

could be : 

`foreach(dir launch)`

and also, you could add a install for the home_robot.py : 

```
catkin_install_python(PROGRAMS scripts/home_robot.py
                      DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}) 
```

**for kortex_example :** 

```
error: package directory 'python/kortex_examples' does not exist
CMake Error at ros_kortex/kortex_examples/catkin_generated/safe_execute_install.cmake:4 (message):
  
  execute_process(/home/roso/Workspaces/kortex/build/ros_kortex/kortex_examples/catkin_generated/python_distutils_install.sh)
  returned error code
Call Stack (most recent call first):
  ros_kortex/kortex_examples/cmake_install.cmake:51 (include)
  cmake_install.cmake:134 (include)
```

and I have no fix for this, I just removed this package since I don't need it but the catkin_make install doesn't work with this one

alexvannobel, 2019-10-21:
Hi @mlauret ,

We did not test the ```catkin_make install``` much, so I'll modify the CMakeLists.txt files to support it. I'll open a PR soon.

Best,
Alex

alexvannobel, 2019-10-28:
Hi @mlauret ,

I merged the PR so now ```catkin_make install``` should work properly.
I'll close this issue, but feel free to reopen it if you still encounter the problem.

Best,
Alex

closed 2019-10-28 by alexvannobel