Kinovarobotics/ros_kortex #51: Namespace mismatch

rrwiyatn, 2019-11-07:
Hi, there seems to be a namespace mismatch. When I tried to run the moveit examples following the instructions in https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_examples/readme.md, I received an error after running `example_move_it_trajectories.py` (https://pastebin.com/6j0P97xQ).

The problem goes away if I do `rosrun kortex_examples example_move_it_trajectories.py __ns:=my_gen3` although I did not change anything in my `kortex_driver.launch`.

alexvannobel, 2019-11-12:
Hi @rrwiyatn ,

I tested a bit with the namespacing mechanics of the MoveIt Python objects, and the best option I found is to explicitely supply the ```__ns``` parameter, but since that's not really clean to do with rosrun, I added a launch file to simplify the whole process of starting the example. 
You can have a look at the documentation [here](https://github.com/Kinovarobotics/ros_kortex/pull/53/commits/9508ce1124c72b7b1aa03316f55e489b2296ee48#diff-99eae0084eab8d421694eba52d6054cfR83). I'll test it on a real robot shortly and then merge the PR to kinetic-devel.

Best,
Alex

alexvannobel, 2019-11-12:
I merged the PR.
Don't hesitate to reopen the issue if you re-encounter this problem.

Best,
Alex

closed 2019-11-12 by alexvannobel

rrwiyatn, 2019-11-16:
Great thanks @alexvannobel!