Kinovarobotics/ros_kortex #52: Robot model parameter not found! Did you remap 'robot_description'?

WHGang, 2019-11-11:
Hello, Thank you for the code.I had an error when I ran the code. After I connect to real robot.
I frist run: roslaunch kortex_driver kortex_driver.launch
Then run: rosrun kortex_examples example_move_it_trajectories.py
An error occurred:
gang@gang-Legion:~$ rosrun kortex_examples example_move_it_trajectories.pyFailed to import pyassimp, see https://github.com/ros-planning/moveit/issues/86 for more info
[ERROR] [1573439723.948360254]: Robot model parameter not found! Did you remap 'robot_description'?
Traceback (most recent call last):
File"/home/gang/catkin_ws/src/ros_kortex/kortex_examples/python/move_it/example_move_it_trajectories.py", line 205, in <module>
    main()
File"/home/gang/catkin_ws/src/ros_kortex/kortex_examples/python/move_it/example_move_it_trajectories.py", line 156, in main
    example = ExampleMoveItTrajectories()
File"/home/gang/catkin_ws/src/ros_kortex/kortex_examples/python/move_it/example_move_it_trajectories.py", line 65, in __init__
    self.robot = moveit_commander.RobotCommander("robot_description")
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/moveit_commander/robot.py", line 151, in __init__
    self._r = _moveit_robot_interface.RobotInterface(robot_description, ns)
RuntimeError: RobotInterfacePython: invalid robot model

Did I do something wrong? Thank you for your advice!

closed 2019-11-12 by WHGang

zehuilu, 2020-02-04:
@WHGang Hi, Did you solve this probelm? I met the same issue today. But it was working correctly.

jeff-hykin, 2023-02-28:
I ran into this issue, but in my case I just needed to restart rviz.

I got more info about it by running `rosparam list` and then looking for "robot_description". Even though I had rviz running it didn't show up on that list until I restarted rviz