Kinovarobotics/ros_kortex #54: The starting point for the trajectory did not match the actual commanded joint angles

truhoang-ubtrobot, 2019-11-15:
I got the following error when trying to execute a pose target using MoveIt! move_group_interface. I have executed the code several with success but sometimes, when I zero the arm manually to the home position (using a joystick) and re-run the code for testing, I get the error. 

```
[ERROR] [1573778245.421304088]: Trajectory has been aborted.
[ERROR] [1573778245.421976633]: Trajectory execution failed in the arm with sub error code 69
The starting point for the trajectory did not match the actual commanded joint angles.
```

Before I plan, I also call setStartStateToCurrentState() to make sure the planner is updated with the current arm's state.

Can provide help to mitigate this issue?

Thanks

gstlaurentkinova, 2019-11-28:
Hi @truhoang-ubtrobot 

Sorry for the late reply.

This is an intermittent issue known to us. Work is currently underway for fixing this issue.

One way to mitigate this issue is to move the robot a little before sending the trajectory. This minimises possible residual error between the starting trajectory point and the commanded joint angle.  

Best,

Gabriel

truhoang-ubtrobot, 2019-12-02:
@gstlaurentkinova 
Do you mean manually moving it a little using the joystick?

Is the fix for this planned for the next release? If so, I'll leave this thread open until then.

gstlaurentkinova, 2020-03-13:
Hi @truhoang-ubtrobot,

> Do you mean manually moving it a little using the joystick?

Yes. This is what I meant.

> Is the fix for this planned for the next release?

It is not planned for the next release. This is problem hard to reproduce on our end.
We are still working on this.

I'll leave this issue open with the `bug` tag and update it whenever we have a fix.

Best,

Gabriel

dniewinski, 2020-04-27:
Hi @truhoang-ubtrobot.  I ran into this issue this morning.  For me, it turned out to be the ordering of the adaptors in the ompl_planning_pipeline.  Make sure "industrial_trajectory_filters/UniformSampleFilter" is the first one in the list and the "sample_duration" parameter is set to 0.001

truhoang-ubtrobot, 2020-04-27:
@dniewinski I'm using default MoveIt settings so the `UniformSampleFilter` is already the first in the list. Thanks for the suggestion

felixmaisonneuve, 2021-06-14:
Hi guys, 

With the release of the version 2.3, modifications in MoveIt configurations have been made that should fix this type of errors. More details about the new release [here](https://github.com/Kinovarobotics/ros_kortex/releases/tag/v2.3.0).

I will keep this issue open so you can try it. I will close it in a couple weeks unless there is still a problem

Regards,
Felix

felixmaisonneuve, 2021-07-26:
I will close this issue.

Please open a new issue or comment/re-open this one if you still get this error

Thank you,
Felix

closed 2021-07-26 by felixmaisonneuve