Kinovarobotics/ros_kortex #55: Joint speed control from ROS

akihikoy, 2019-11-22:
As mentioned in [kortex-issues#3](https://github.com/Kinovarobotics/kortex/issues/3), the [SendJointSpeedsCommand](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/summary_pages/Base.md) function is now available with the 2.0 release.
Can we control joint speed from ROS?  In that case, could you explain how to do that?
Or, should we directly use `SendJointSpeedsCommand` through `kortex`?

Best,
--Akihiko

akihikoy, 2019-11-26:
I have found the way of joint velocity control of Gen3.
On ROS, [SendJointSpeedsCommand](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/summary_pages/Base.md) is available through the `/my_gen3/base/send_joint_speeds_command` service.  It seems to be a standard joint angular velocity control, and there is no issue so far.
A note is that we need to specify the angular velocities in **degrees** per second (not rad per sec as usual ROS message).

My test code is available from: https://github.com/akihikoy/ay_test/blob/master/ros/py_ros/gen3/velctrl1.py
Video: https://youtu.be/BzuPKjanXP4

gstlaurentkinova, 2019-11-26:
Hi @akihikoy 

Glad you were able to solve your issue. This is indeed the preferred solution.

I will close this issue since the problem seems to be solved.

Do not hesitate to reopen this issue if there is anything with the functionality.

Best,
Gabriel

closed 2019-11-26 by gstlaurentkinova