Kinovarobotics/ros_kortex #57: cyclic_data_publish_rate does not change the joint_states publishing rate

akihikoy, 2019-11-25:
Hello,
I'm trying to change the publishing rate of `joint_states`.

According to the [kortex_driver/readme.md](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/readme.md), `cyclic_data_publish_rate` is the parameter for changing that, and it's default is 100.
However, when I launch the drive by:
```
$ roslaunch kortex_driver kortex_driver.launch ip_address:=gen3a start_moveit:=false start_rviz:=true robot_name:=gen3a
```
the average rate of `/gen3a/joint_states` is around 10.
```
$ rostopic hz /gen3a/joint_states
subscribed to [/gen3a/joint_states]
average rate: 10.001
        min: 0.099s max: 0.101s std dev: 0.00043s window: 10
average rate: 10.001
        min: 0.099s max: 0.101s std dev: 0.00040s window: 20
average rate: 10.000
        min: 0.099s max: 0.101s std dev: 0.00037s window: 30
```

I added `cyclic_data_publish_rate:=200` and then `cyclic_data_publish_rate:=5` at the end of the launch command line, but it had no effect.  Although the SUMMARY of the launch showed the parameter was changed,
```
PARAMETERS
...
 * /gen3a/gen3a_driver/cyclic_data_publish_rate: 5
```
the rate measured by `rostopic hz` was still 10.

If this issue is due to a kind of latency, it should be able to decrease the rate under 10, but even if setting the rate to be 5, the actual rate is 10.
This means that I am misunderstanding, or there is a bug in `kortex_driver`.
Any ideas?

akihikoy, 2019-11-25:
Updates:

The topics `/gen3a/base_feedback` and `/gen3a/base_feedback/joint_state` seem to have the rate specified by `cyclic_data_publish_rate`.

However it is strange again, since in the [kortex_driver/readme.md](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/readme.md), there are descriptions:

-

akihikoy, 2019-11-25:
Okay, I found the issue.  It was due to missing `rate` parameter of the `joint_state_publisher` node.  Its default value is 10, regardless its source (`base_feedback/joint_state`).

Fixing this issue is easy.  In [kortex_driver.launch](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/launch/kortex_driver.launch), just specify the `rate` parameter of the node `joint_state_publisher`.  e.g.
```
        <!-- Start joint and robot state publisher -->
        <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
            <param name="rate" value="$(arg cyclic_data_publish_rate)" />
            <rosparam param="source_list">[base_feedback/joint_state]</rosparam>
        </node>
```
(Only the 3rd line was added.)

alexvannobel, 2019-11-25:
Hi @akihikoy ,

You're right, we missed that parameter! I'll make a pull request to fix it.

Thanks,
Alex

alexvannobel, 2019-11-25:
The PR is merged, so I'll close this issue. 
Don't hesitate to reopen it if you re-encounter this problem.

Best,
Alex

closed 2019-11-25 by alexvannobel