Kinovarobotics/ros_kortex #59: Joint control with kortex

Hongjun1108, 2019-11-27:
In kortex_driver.launch, you provided a method to control kortex with move_it, but I want to control in in joint space, just send command to each joint. Then, how can I achieve this?

gstlaurentkinova, 2019-12-02:
Hi @Hongjun1108 ,

Control in joint space can be accomplished in multiple ways:

1) You can send Joint angle commands (commanding the position of the actuators) using the kortex_driver node. Look at the PlayJointTrajectory service in ros_kortex/kortex_driver/srv/generated/base/. Look at the ConstrainedJointAngles command type message located in ros_kortex/kortex_driver/msg/generated/base/. Calling the service should move the robot to the specified position. This is a high-level function for sending a position goal.

2) You can send Joint speed commands (commanding the speed of the actuators) using the kortex_driver node. Look at the SendJointSpeedsCommand and SendSelectedJointSpeedCommand services in ros_kortex/kortex_driver/srv/generated/base/. This service receives a JointSpeeds message, located in ros_kortex/kortex_driver/msg/generated/base/. It will effectively set the actuator speeds as specified.

3) You can precompute a trajectory and send it to the robot to execute. This is the method used in MoveIt. MoveIt will compute a trajectory based on the current robot pose and the desired pose, and send it directly to the robot. There is an utility for this located in ros_kortex/kortex_driver/include/kortex_driver/non-generated/pre_computed_joint_trajectory_action_server.h and  ros_kortex/kortex_driver/src/kortex_driver/non-generated/pre_computed_joint_trajectory_action_server.cpp that subscribes to the gen3_joint_trajectory_controller/follow_joint_trajectory action and executes a precomputed joint trajectory when supplied with one.

Tell me if this answers your question.

Best,
Gabriel

alexvannobel, 2020-01-28:
Hi @Hongjun1108 ,

I'll close this issue for inactivity, but feel free to re-open it if it persists or to open a new issue.

Best regards,
Alex

closed 2020-01-28 by alexvannobel