Kinovarobotics/ros_kortex #60: Difference of EEF position between kortex_driver and MoveIt

juntuck, 2019-11-28:
Greetings,

there exists an offset between the tool/end effector pose returned by kortex's BaseCyclic_Feedback.pose.tool_pose_xyz and MoveIt's arm_group.get_current_pose().

For example with the EEF facing towards the ground:

**MoveIt:**
    x: -0.0237346354686
    y: 0.24785572974
    z: 0.0580734261844
  orientation: 
    x: -0.0432058638949
    y: -0.99869048994
    z: -0.0215183862723
    w: 0.0169563463014

**kortex_driver:**
  position: 
    x: -0.0275755301118
    y: 0.253189027309
    z: -0.0617465749383
  orientation: 
    x: 0.043206269999861865
    y: 0.9986904800693364
    z: 0.021518075282811973
    w: -0.01695628755655937

Difference:
  x: 0.00384089464324
  y: -0.00533329756901
  z: 0.119820001123

Upon testing some other poses, it looks like MoveIt defines the EEF link ~10cm further out from the bracelet joint than kortex_driver does.

This difference in readings is problematic, as I would like to freely switch between controlling the arm with kortex_driver and MoveIt. What would you suggest?

Best regards,
Jude

alexvannobel, 2019-12-02:
Hi @juntuck ,

This seems related to #26 , so I'll answer on that issue.

Best, 
Alex

juntuck, 2019-12-10:
Dear @alexvannobel ,

We found the reason for this :) It is because in the Web Interface, under Configuration -> Tool -> Robot, there was a 0.12m offset in the z-axis defined for the tool.

alexvannobel, 2019-12-10:
Hi @juntuck ,

If you put a "0.00m" tool offset in the WebApp, do MoveIt and the kortex_driver return the same tool poses now?

juntuck, 2019-12-11:
Yup they do! :)

alexvannobel, 2020-06-09:
Hi @juntuck ,

I finally found the source of this. The arm's firmware has some hardcoded values for end-effectors and tool frames. I added those frames to the xacro's and the `tool_frame` link should now correctly reflect the robot's feedback. It was also added to the `arm` move_group in MoveIt for every arm type.

I'll close the issue now but don't hesitate to reopen if you have questions!
Best,
Alex

closed 2020-06-09 by alexvannobel

elevenjiang1, 2021-01-31:
Sorry,I also find that the end pose from  BaseCyclic_Feedback.pose is also different to TF tree(end_effector_link).But the web interface already set tool Transfomr to all 0.
(end_effector_link is same as tool_frame)

elevenjiang1, 2021-02-07:
> Sorry,I also find that the end pose from BaseCyclic_Feedback.pose is also different to TF tree(end_effector_link).But the web interface already set tool Transfomr to all 0.
> (end_effector_link is same as tool_frame)

Alright...After check ,I find that BaseCyclic_Feedback is accuracy.Xacro need to add a static transform.

```
    <!-- Link Kinova True end and Xacro  -->
    <link name="true_effector_link"/>
    <joint name="robot_accuracy_joint" type="fixed">
        <parent link="end_effector_link" />
        <child link = "true_effector_link" />
        <origin xyz="0.002 0.005 0.004" rpy="-0.007 0.006 -0.009" />
    </joint>
```

elevenjiang1, 2021-02-09:
After test for a long...,end_effector and feedback is not a fixed transform....(0.1mm different).But this make my project can not work!!
So hope to solve it!

elevenjiang1, 2021-02-09:
After test for a long...,end_effector and feedback is not a fixed transform......(max nearly 3mm different!!)
So hope to solve it!