Kinovarobotics/ros_kortex #63: control  problems

WHGang, 2019-12-11:
I saw a view(https://www.youtube.com/watch?v=8MTuTNIQUqA) on YouTube.  At eht end of the view, gem3 can realize follow the goal in real time. I want to know how is it controlled?  Thanks!

WHGang, 2019-12-13:
[image]
just like jaco2!

alexvannobel, 2020-01-28:
Hi @WHGang ,

I just merged #72 that enables sending joint speed and twist as a topic to the arm. You will find the details in the [readme](https://github.com/Kinovarobotics/ros_kortex/tree/kinetic-devel/kortex_driver#robot-control-topics-in).

I'll close this issue, but feel free to reopen it if you experience problems wiht it. 

Best regards, 
Alex

closed 2020-01-28 by alexvannobel