Kinovarobotics/ros_kortex #66: Moveit example doesn't work on real manipulator

XuRuoyuRoy, 2019-12-28:
Hello, I have a problem on this manipulator. The ROS example of joint control and cartesian control works well on the real robot. The moveit example in the kortex_example package also works well in Gazebo, however, when it comes to the real robot, all the commands are rejected. The error shows as follow: 
[ INFO] [1577536720.839343151]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1577536720.857312273]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1577536720.904154537]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1577536720.907537266]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1577536720.907631471]: Solution found in 0.003769 seconds
[ INFO] [1577536720.925600314]: SimpleSetup: Path simplification took 0.017706 seconds and changed from 4 to 2 states
[ INFO] [1577536720.933210109]: Execution request received
[ INFO] [1577536720.939686916]: New goal received.
[ERROR] [1577536720.939746196]: Joint Trajectory Goal is rejected.
[ WARN] [1577536720.940266595]: Controller handle gen3_joint_trajectory_controller reports status FAILED
[ INFO] [1577536720.940329222]: Completed trajectory execution with status FAILED ...
[ INFO] [1577536720.940455986]: Execution completed: FAILED
Can you help me fix it?

dniewinski, 2020-01-02:
Seeing exactly the same issue.  When digging a little harder, looks like the paths created from moveit aren't detailed enough, so the arm is rejecting them:

`-----------------------------
Error #1
Type : TRAJECTORY_ERROR_TYPE_INVALID_TIME_STEP
Identifier : TRAJECTORY_ERROR_IDENTIFIER_TIME
Actuator : 1
Erroneous value is 0.191003 but minimum permitted is 0.001 and maximum permitted is 0.001
Additional message is : Invalid Time Step - Time step between Trajectory Point (1) and previous one d
iffers from predefined Time Step
-----------------------------
-----------------------------
Error #2
Type : TRAJECTORY_ERROR_TYPE_INVALID_POSITION
Identifier : TRAJECTORY_ERROR_IDENTIFIER_POSITION
Actuator : 2
Erroneous value is 1.0502 but minimum permitted is -0.050002 and maximum permitted is 0.050002
Additional message is : Invalid Position - Position variation between Trajectory Point (1) and previo
us one unacheivable with current speed limits
-----------------------------
`

dniewinski, 2020-01-02:
So, i solved my own problem.  In my moveit config, I didn't include the uniform sample filter. https://github.com/Kinovarobotics/ros_kortex/blob/cb95593c52f91f2e444afb3873111361772d9ffa/kortex_move_it_config/gen3_robotiq_2f_85_move_it_config/launch/ompl_planning_pipeline.launch.xml#L8

alexvannobel, 2020-01-06:
Hi @XuRuoyuRoy ,

As @dniewinski pointed out, we use the uniform sample filter in our MoveIt configurations because the arm accepts trajectories with 1ms timesteps only for now.

If you try it with the filter, does it work on the real robot?

Best,
Alex

XuRuoyuRoy, 2020-01-08:
> Hi @XuRuoyuRoy ,
> 
> As @dniewinski pointed out, we use the uniform sample filter in our MoveIt configurations because the arm accepts trajectories with 1ms timesteps only for now.
> 
> If you try it with the filter, does it work on the real robot?
> 
> Best,
> Alex

Hello @alexvannobel 

I tried this today, and delet the "industrial_trajectory_filters/UniformSampleFilter" in both "ros_kortex/kortex_move_it_config/gen3_robotiq_2f_85_move_it_config" and "ros_kortex/kortex_move_it_config/gen3_move_it_config", but the same error is still exist. It seems we have different problems.

Best,
Roy

alexvannobel, 2020-01-28:
Hi @XuRuoyuRoy ,
Sorry for the delay.

You will have to keep the uniform sample filter, as the arm only accepts points with 1ms timesteps and this is the simple way to interpolate them. If you keep the filter in both "ros_kortex/kortex_move_it_config/gen3_robotiq_2f_85_move_it_config" and "ros_kortex/kortex_move_it_config/gen3_move_it_config", and launch the example, what is the error as displayed in the Kortex Driver console?

Also, did you do a rosdep update with ros_kortex packages? Maybe you don't have the industrial_sample_filters package installed. 

Let me know,
Best,
Alex

alexvannobel, 2020-03-10:
Hi,

I will close this issue for inactivity and because right now, there is no way to bypass the industrial_sample_filters because the arm needs points sampled at 1ms.
Don't hesitate to reopen if you have any more questions.

Best,
Alex

closed 2020-03-10 by alexvannobel

bradley-whitlock, 2020-03-11:
I am continuing to get this error on my Gen3 with "Joint Trajectory Goal is rejected." when generating a path with **compute_cartesian_path()**. Is there any easy way to fix the sampling time at 1ms.

huiwenzhang, 2020-06-29:
@dniewinski Still get the error
> Error #3
Type : TRAJECTORY_ERROR_TYPE_INVALID_ACCELERATION
Identifier : TRAJECTORY_ERROR_IDENTIFIER_ACCELERATION
Actuator : 1
Erroneous value is 52.3602 but minimum permitted is 0 and maximum permitted is 51.5662
Additional message is : Invalid Acceleration - Acceleration of actuator(0) in Trajectory Point (1490) exceeds limits

> Error #4
Type : TRAJECTORY_ERROR_TYPE_INVALID_ACCELERATION
Identifier : TRAJECTORY_ERROR_IDENTIFIER_ACCELERATION
Actuator : 1
Erroneous value is 52.6287 but minimum permitted is 0 and maximum permitted is 51.5662
Additional message is : Invalid Acceleration - Acceleration of actuator(0) in Trajectory Point (1491) exceeds limits

What is going on here? I am using the default setting of the code, as pasted below
https://github.com/Kinovarobotics/ros_kortex/blob/a57ff4ce4336bd5c4fa997e85734ac19399cece1/kortex_move_it_config/gen3_move_it_config/launch/ompl_planning_pipeline.launch.xml#L6-L13

dniewinski, 2020-06-29:
The only way I found to work around this was to change the URDF to have MUCH smaller acceleration and speed values.  Arm moves slowly, but at least it works.  This isn't the correct solution, but will let you move forward at least

huiwenzhang, 2020-07-10:
> The only way I found to work around this was to change the URDF to have MUCH smaller acceleration and speed values. Arm moves slowly, but at least it works. This isn't the correct solution, but will let you move forward at least

Hi, thanks for your reply. I did as you suggested, changed the soft velocity and acceleration limits to be 0.3 and 0.1 (the default value is 0.82 and 0.3), but it doesn't work or works occasionally. I guess the logic behind this is smaller vel and acc  limits configurations make the MoveIt output trajectory with much lower vel  and acc, and thus has less probability to violate the hardware constraints. 
What I did is computing the trajectory with MoveIt and then call the `PrecomputedJointTrajectoryAction` to execute the trajectory. I checked the trajectory, they did have 1ms timestep interval, but the controller still rejects to execute it. I think this bug is caused by the `playJointTrajectory` method implemented in the Kotex API. What's your opinion? Any suggestions to solve this?

elevenjiang1, 2021-01-28:
> > The only way I found to work around this was to change the URDF to have MUCH smaller acceleration and speed values. Arm moves slowly, but at least it works. This isn't the correct solution, but will let you move forward at least
> 
> Hi, thanks for your reply. I did as you suggested, changed the soft velocity and acceleration limits to be 0.3 and 0.1 (the default value is 0.82 and 0.3), but it doesn't work or works occasionally. I guess the logic behind this is smaller vel and acc limits configurations make the MoveIt output trajectory with much lower vel and acc, and thus has less probability to violate the hardware constraints.
> What I did is computing the trajectory with MoveIt and then call the `PrecomputedJointTrajectoryAction` to execute the trajectory. I checked the trajectory, they did have 1ms timestep interval, but the controller still rejects to execute it. I think this bug is caused by the `playJointTrajectory` method implemented in the Kotex API. What's your opinion? Any suggestions to fix this?

Hello,I also face with this problem.Now kinova can only move very very slow...(0.03m/s for end_effector).Any method to slove this problem now?
(I want to use FollowJointTrajectory for I need to use UR5 and Kinova,I think maybe moveit can help me to slove this problem,but my interface will change a lot...)

Alchemist77, 2024-10-29:
**solution**: in moveit folder, search default_joint_limits.yaml
in the document, search joint_limits and then
  reduce  max_acceleration,  from max_acceleration: 0.84 ->     max_acceleration: 0.5 for error joint and then the error will be disappeared.