Kinovarobotics/ros_kortex #68: New update PR #65 fail to build

truhoang-ubtrobot, 2019-12-31:
Hello,

I `git pull` the latest change [PR 65](https://github.com/Kinovarobotics/ros_kortex/pull/65) to test the new updates on the controller. I got the following errors: 

```

/home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/kortex_driver/src/generated/base_services.cpp: In member function ‘bool BaseServices::UpdateMapping(kortex_driver::UpdateMapping::Request&, kortex_driver::UpdateMapping::Response&)’:
/home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/kortex_driver/src/generated/base_services.cpp:1098:11: error: ‘class Kinova::Api::Base::BaseClient’ has no member named ‘UpdateMapping’; did you mean ‘CreateMapping’?
   m_base->UpdateMapping(input, m_current_device_id, m_api_options);
           ^~~~~~~~~~~~~
           CreateMapping
/home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/kortex_driver/src/generated/base_services.cpp: In member function ‘bool BaseServices::DeleteMapping(kortex_driver::DeleteMapping::Request&, kortex_driver::DeleteMapping::Response&)’:
/home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/kortex_driver/src/generated/base_services.cpp:1130:11: error: ‘class Kinova::Api::Base::BaseClient’ has no member named ‘DeleteMapping’; did you mean ‘CreateMapping’?
   m_base->DeleteMapping(input, m_current_device_id, m_api_options);
           ^~~~~~~~~~~~~
           CreateMapping
/home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/kortex_driver/src/generated/base_services.cpp: In member function ‘bool BaseServices::ReadMap(kortex_driver::ReadMap::Request&, kortex_driver::ReadMap::Response&)’:
/home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/kortex_driver/src/generated/base_services.cpp:1232:20: error: ‘class Kinova::Api::Base::BaseClient’ has no member named ‘ReadMap’; did you mean ‘ReadMapping’?
   output = m_base->ReadMap(input, m_current_device_id, m_api_options);
                    ^~~~~~~
                    ReadMapping
/home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/kortex_driver/src/generated/base_services.cpp: In member function ‘bool BaseServices::UpdateMap(kortex_driver::UpdateMap::Request&, kortex_driver::UpdateMap::Response&)’:
/home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/kortex_driver/src/generated/base_services.cpp:1265:11: error: ‘class Kinova::Api::Base::BaseClient’ has no member named ‘UpdateMap’; did you mean ‘CreateMap’?
   m_base->UpdateMap(input, m_current_device_id, m_api_options);
           ^~~~~~~~~
           CreateMap
/home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/kortex_driver/src/generated/base_services.cpp: In member function ‘bool BaseServices::DeleteMap(kortex_driver::DeleteMap::Request&, kortex_driver::DeleteMap::Response&)’:
/home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/kortex_driver/src/generated/base_services.cpp:1297:11: error: ‘class Kinova::Api::Base::BaseClient’ has no member named ‘DeleteMap’; did you mean ‘CreateMap’?
   m_base->DeleteMap(input, m_current_device_id, m_api_options);
           ^~~~~~~~~
           CreateMap
ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/build.make:230: recipe for target 'ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/src/generated/base_services.cpp.o' failed
```

Since the package build successfully with the previous version, I suspect the error is not a result of anything from my end.

alexvannobel, 2020-01-06:
Hi @truhoang-ubtrobot ,

Can you try deleting the kortex_api/include and kortex_api/lib folders, removing the build and devel folders from the base of the workspace and rebuilding again? I suspect this may be a mismatch between the Kortex API and the ROS driver, which will be solved when the download script downloads the latest API after deleting it locally. 

Tell me how this goes,
Best, 
Alex

truhoang-ubtrobot, 2020-01-06:
@alexvannobel,

I did that and got the following: 
```
In file included from /home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/kortex_driver/src/generated/actuatorconfig_services.cpp:17:0:
/home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/kortex_driver/include/kortex_driver/generated/common_proto_converter.h:28:10: fatal error: Common.pb.h: No such file or directory
 #include <Common.pb.h>
          ^~~~~~~~~~~~~
compilation terminated.
ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/build.make:110: recipe for target 'ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/src/generated/actuatorconfig_services.cpp.o' failed
make[2]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/src/generated/actuatorconfig_services.cpp.o] Error 1
In file included from /home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/kortex_driver/src/generated/actuatorconfig_ros_converter.cpp:17:0:
/home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/kortex_driver/include/kortex_driver/generated/actuatorconfig_ros_converter.h:28:10: fatal error: ActuatorConfig.pb.h: No such file or directory
 #include <ActuatorConfig.pb.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/build.make:86: recipe for target 'ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/src/generated/actuatorconfig_ros_converter.cpp.o' failed
make[2]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/src/generated/actuatorconfig_ros_converter.cpp.o] Error 1
In file included from /home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/kortex_driver/src/generated/actuatorcyclic_proto_converter.cpp:17:0:
/home/truhoang/1-Project_Code/ubtech_ws/src/ros_kortex/kortex_driver/include/kortex_driver/generated/actuatorcyclic_proto_converter.h:28:10: fatal error: ActuatorCyclic.pb.h: No such file or directory
 #include <ActuatorCyclic.pb.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/build.make:134: recipe for target 'ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/src/generated/actuatorcyclic_proto_converter.cpp.o' failed
make[2]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/src/generated/actuatorcyclic_proto_converter.cpp.o] Error 1
CMakeFiles/Makefile2:5097: recipe for target 'ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/all' failed
make[1]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4" failed
The terminal process terminated with exit code: 1
```

alexvannobel, 2020-01-06:
Hi @truhoang-ubtrobot ,

I just tried these two steps with the HEAD of the kinetic-devel branch (delete build and devel folders, as well as kortex_api/include and kortex_api/lib folders) and `catkin_make` worked without fail.

Your error seems related to the Kortex API not being found. 

A couple things to check:
- Verify you are on latest commit of the kinetic-devel branch.
- Can you run `cd kortex_api/ && ls` to make sure the API has been downloaded correctly? 
- If you `ls kortex_api/include`, what is inside include?
- Can you check upon launching `catkin_make` if you see the download has failed? You will see this in the log if the download is a success: 

```
-- +++ processing catkin package: 'kortex_driver'
-- ==> add_subdirectory(kortex_driver)
include does not exist, creating directory...
lib does not exist, creating directory...
Downloading the Kortex API from the Web...
[Info] Searching items to download...
[Info] [Thread 2] Downloading generic-local-stable/API/2.1.0/kortex_api_2.1.0-3.zip
[Info] [Thread 2] [0]: 206 Partial Content...
[Info] [Thread 2] [2]: 206 Partial Content...
[Info] [Thread 2] [1]: 206 Partial Content...
[Info] [Thread 2] Done downloading.
{{
  "status": "success",
  "totals": {{
    "success": 1,
    "failure": 0
  }}
}}
```
Let me know how this goes,
Best,
Alex

truhoang-ubtrobot, 2020-01-07:
@alexvannobel,

I checked the `include` folder and indeed no headers were downloaded. I went into the merge from PR #65 and checked the bash script (`download_kortex_api.bash`). The following line is the reason:
```
jfrog rt dl --flat  generic-local-stable/API/2.1.0/kortex_api_2.1.0-3.zip ./
```

Since I don't already have `jfrog` I installed it. So now `jfrog` is visible to the OS then I got the following message:
```
-- ==> add_subdirectory(ros_kortex/kortex_driver)
include does not exist, creating directory...
lib does not exist, creating directory...
Downloading the Kortex API from the Web...
To avoid this message in the future, set the JFROG_CLI_OFFER_CONFIG environment variable to false.
The CLI commands require the Artifactory URL and authentication details
Configuring JFrog CLI with these parameters now will save you having to include them as command options.
You can also configure these parameters later using the 'config' command.

Configure now? (y/n): mv: cannot stat 'kortex_api_2.1.0-3.zip': No such file or directory
unzip:  cannot find or open kortex_api.zip, kortex_api.zip.zip or kortex_api.zip.ZIP.
ERROR while extracting the kortex api. code = 9

```

alexvannobel, 2020-01-07:
Hi @truhoang-ubtrobot ,
You're right, the download script was not completely up-to-date. Sorry for the inconvenience.
I pushed the fix onto the kinetic-devel branch and I will update the 2.1.0 tag to use this fix instead. 
I now use `wget` to fetch the archive and so it should work.
Let me know how this goes,
Best,
Alex

truhoang-ubtrobot, 2020-01-07:
@alexvannobel,

Build successful! Thanks.

closed 2020-01-07 by truhoang-ubtrobot