Kinovarobotics/ros_kortex #69: Request for Ros 2 Support

anthonytec2, 2020-01-03:
ROS 2 support for the kortex api

alexvannobel, 2020-01-28:
Hi @anthonytec2 ,

I don't have a date to give you but we consider adding ROS2 support. I'll keep this issue updated as I have more info to give you.

Thanks for the feedback,
Alex

ngeerlingsriwo, 2021-02-16:
Hi @alexvannobel,

Are there any updates regarding ROS2 support?

felixmaisonneuve, 2021-06-14:
Hi @ngeerlingsriwo,

There is still no change regarding ROS2. 
We are prioritizing moving to ROS Neotic first (we are currently using ROS Melodic), but we still consider adding ROS2 support.

Best,
Felix

civerachb-cpr, 2021-09-14:
Since it appears there's a Noetic branch available now, is there an updated timeline on when we might see ROS2 support?  Or a definitive "we've considered it, but have decided against it" answer?

civerachb-cpr, 2022-09-21:
It's almost a year later. Are we any closer to a timeline for ROS2 support?

felixmaisonneuve, 2022-09-22:
Hi @civerachb-cpr,
We can confirm a ROS2 implementation will be done for Gen3, but do not have a release timeline yet.

I will notify that clients are asking for this and I can recommand you to contact your Kinova sales representative (since you are from Clearpath Robotics) and ask for this too. 
This is not my decision, but with pressure things might start moving faster. 

Regards,
Felix

ngeerlingsriwo, 2022-10-28:
Will Ubuntu 22.04/ROS2 Humble LTS be supported in the ROS2 implementation?

felixmaisonneuve, 2023-05-18:
ROS2 support will be for Ubuntu22.04

felixmaisonneuve, 2023-08-17:
ROS2 support is now available for Ubuntu22.04.
You can find it here : https://github.com/Kinovarobotics/ros2_kortex

Closing this issue

closed 2023-08-17 by felixmaisonneuve