Kinovarobotics/ros_kortex #70: Is the low-level control services available in the default kortex_driver package?

XuRuoyuRoy, 2020-01-10:
I want to do low level control. but I can't find any services in the source code of the kortex_driver package. So is that available by default? If it is not generated in the provided code, is there any tutorial or example to state it more clear and in detail?

gstlaurentkinova, 2020-02-27:
Hi @XuRuoyuRoy,

Low-level control is not available through the ROS1 interface because of real-time issues. Currently the ROS1 implementation does not integrate real-time capabilities with the Gen3 arms.

As such, the services for low-level control are disabled in the kortex_driver package. I suggest you look into the Api examples (https://github.com/Kinovarobotics/kortex) for low-level examples using the Kortex Api.

I hope this answers your question.

Best,

Gabriel

polthergeist, 2020-04-07:
Hi @XuRuoyuRoy, I was wondering, fo you have any other question regarding this thread or we can close it?

XuRuoyuRoy, 2020-04-08:
No，feel free to close that， thanks for your help?.

closed 2020-04-08 by polthergeist