Kinovarobotics/ros_kortex #71: Best way to do continuous and smooth Cartesian position control?

jloboprat, 2020-01-16:
We are trying to teleoperate the Kinova Gen3 robot with a motion capture system. We have tried sending target poses but the resulting motion is very jerky.

What would be the best way to achieve continuous and smooth Cartesian position control?

RaduCorcodel, 2020-01-16:
Are you sending via points to the position control service? That would make the motion jerky because the robot decelerates and accelerates at each via point.
If so, one way to do it is to use the velocity controller service and you specify velocity vectors from your current configuration to the next via point. I believe this would work best especially if you don't know the via points in advance so the robot would constantly chase after a "moving target". 
If you know all the via points in advance, then you can spline them with a 1ms time interval in between them and send them to the pre-computed trajectory controller. 
All robots that we used had this functionality available in a ros_control package (supplied straight by the manufacturer). I, and many other users, have reached Kinova on _multiple_ occasions to give us such functionality but it seems they don't find it all that important...

polthergeist, 2020-04-07:
Hi @jloboprat, I agree with @RaduCorcodel, for now the best way to achieve a smooth movement is via the velocity control. However, before the COVID-19 situation, the control team was working on a new waypoint system that can allow the kind of behaviour that you are looking for. Do you want to keep this thread open until then or we can close it?

jloboprat, 2020-04-08:
Hi @polthergeist, I would prefer to keep this thread open until we can test the new waypoint system that you mention. Thanks!

polthergeist, 2020-04-08:
Excellent, no problem!
I'll keep you posted.

lingxiaomeng, 2021-03-23:
Hi @RaduCorcodel, I just found that waypoints can not be used. May I ask if there is an example for using velocity controller service used. Thanks!

polthergeist, 2021-03-23:
Hi @lingxiaomeng I did not see anything on the ROS repo but there is one example on the kortex repo. https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/102-Movement_high_level/03-twist_command.cpp
It gives an idea on how to use twist command. I think you can easily translate it to make it work on your ROS application. Don't hesitate if this does not help.

RaduCorcodel, 2021-03-23:
Hi @lingxiaomeng, @jloboprat,

There is no example in kortex_examples for this, but I can show you how I did it for our Kinova robots. It was just a quick&dirty way to test it out and I'm sure you can massage it to make it work nicer.

Let's say that you want the robot to continuously chase a moving target. That is, from the current pose, you want to command the robot in velocity mode to go toward a configuration in joint space. Then you would do something like this:

First you calculate a vector in joint space between current configuration and goal configuration. I'm doing all computations in joint space but if you prefer Cartesian space, just use KDL to calculate the inverse kinematics.
```
// Get vector distance in joint space between current state and goal
std::vector<double> VecDifferenceGoal2Actual(const trajectory_msgs::JointTrajectoryPoint& p1) 
{
    int nDOF = baseFeedback.actuators.size();
    double d2r = M_PI/180;
    ROS_ASSERT(nDOF == p1.positions.size()); //Complain if the goal has different # of DOFs than what the robot has
    std::vector<double> res(nDOF);
    double jn1, jn0;
    for (size_t i=0; i<nDOF; i++) {
        jn1 = p1.positions[i];
        jn0 = baseFeedback.actuators[i].position*d2r;
        res[i] = jn1 - jn0;
    }
    return res;
}
```

Then you prepare a velocity command:
```
kortex_driver::SendJointSpeedsCommand ComputeVelocityCommand(
                                        double* dist, 
                                        const trajectory_msgs::JointTrajectoryPoint& goal)
{
    int nDOF = goal.positions.size();
    double d2r = M_PI/180; 

    std::vector<double> vecDiffGoalToCurrent = VecDifferenceGoal2Actual(goal);

    double totalAngularVelocity = ComputeVectorNorm(vecDiffGoalToCurrent); //This is just the norm of an n-d vector. Pretty self explanatory

    kortex_driver::SendJointSpeedsCommand jointVelCmd;
    kortex_driver::JointSpeed jointSpeed;
    float speedMultiplier = 15.0f; // This controls how fast the robot will move. Better implement as a parameter
    for (size_t i=0; i<nDOF; i++) {
        jointSpeed.joint_identifier = i;
        jointSpeed.value = (vecDiffGoalToCurrent[i]/totalAngularVelocity) * speedMultiplier;
        jointVelCmd.request.input.joint_speeds.push_back(jointSpeed);
    }
    *dist = totalAngularVelocity;
    return jointVelCmd;
}
```

Now your loop sends commands to the kortex service:
```
// Initialize the ServiceClient
ros::ServiceClient serviceClientJointSpeeds = nh.serviceClient<kortex_driver::SendJointSpeedsCommand>("/my_gen3/base/send_joint_speeds_command");
double vecDiffLength;
double goal_tolerance = 4.0e-2;

// You must populate this with the target that you want the robot to go to (may be one point or a via point from a list)
trajectory_msgs::JointTrajectoryPoint trjPoint; 

while (vecDiffLength > goal_tolerance) {
  kortex_driver::SendJointSpeedsCommand jointVelocityCmd;
  jointVelocityCmd = ComputeVelocityCommand(&vecDiffLength, trjPoint);
  if (!serviceClientJointSpeeds.call(jointVelocityCmd)) {
    ROS_ERROR("Problem sending velocity command");
  }
  ros::Duration(0.1).sleep();
}

//When you're done, make sure you send a zero velocity command otherwise the robot will continue moving slightly around the target
kortex_driver::SendJointSpeedsCommand zeroJointVelocityCmd;
for (size_t i = 0; i < nDOF; i++) {
  kortex_driver::JointSpeed zeroJointSpeed;
  zeroJointSpeed.joint_identifier = i;
  zeroJointSpeed.value = 0.0f;
  zeroJointVelocityCmd.request.input.joint_speeds.push_back(zeroJointSpeed);
}
serviceClientJointSpeeds.call(zeroJointVelocityCmd);
ROS_INFO("Motion should've stopped");
```

You can implement this as a service perhaps and keep giving it via points one by one as you're approaching the previous via point. I trimmed quite a bit from my working code so if I missed anything let me know and I'll revise this post so other people can use it.

alexvannobel, 2021-06-04:
Hi @jloboprat ,

Release 2.3.0 adds a Cartesian trajectory action server, which executes a smooth Cartesian trajectory.

You have to know all your via-points in advance and then send them to the server though, so you cannot do continuous control with it.

Hope this helps,
Best,
Alex

felixmaisonneuve, 2021-07-26:
Hi @jloboprat,

Since this new feature is now released, I will close this issue.

Thank you,
Felix

closed 2021-07-26 by felixmaisonneuve