Kinovarobotics/ros_kortex #77: Joint torque control with Kinova

Hongjun1108, 2020-02-02:
Hi all,
I have kinova gen3, and I want to control it using ROS through joint torque command, however I did not see any examples. I have tried example_full_arm_movement.cpp, and realized joint position and velocity controls. Did anyone know how to do joint torque control?

Thanks a lot.
Hongjun

gstlaurentkinova, 2020-02-27:
Hi @Hongjun1108,

Torque commands are not available through the ROS1 interface, as they are restricted to low-level (real-time) Api. 

I suggest you look into the Kortex examples (https://github.com/Kinovarobotics/kortex) for torque-sepcific examples (https://github.com/Kinovarobotics/kortex/tree/master/api_python/examples/108-Gen3_torque_control and https://github.com/Kinovarobotics/kortex/tree/master/api_cpp/examples/108-Gen3_torque_control in Python and C++, respectively). As of the current version (2.2.0), torque control in ROS1 is not implemented.

Hope this answers your question.

Best,

Gabriel

kaushikbalasundar, 2022-03-14:
Can anyone confirm if this is implemented yet?

felixmaisonneuve, 2022-03-21:
Hi @kaushikbalasundar 

This feature is still not implemented as of now and is not on the roadmap in the short term

Best,
Felix

TJMolnar, 2022-10-13:
I would also love to see this feature supported, as we are looking at using the Gen3 for applications where torque control is desirable. It is certainly possible to run a 1kHz control loop with ROS1 interfaces, as an example Franka has a guide on setting up a system for 1KHz control (and in fact it is a mandatory requirement for running their control interface at any application level): https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel.

sinaaghli, 2023-01-11:
With much delay, there seems to be a solution!
https://github.com/Kinovarobotics/kortex/blob/392c312d57912e9645c70cb3e783bdcdb89371bd/api_cpp/examples/108-Gen3_torque_control/01-torque_control_cyclic.cpp#L266

felixmaisonneuve, 2023-01-11:
I do not want to disapoint you @sinaaghli, but this example has been available for a long time now. It is linked in the second comment of this thread.

However, this feature is not available throught ROS1 because it requires the arm to be in low-level control. As of now, it is still not possible to control the arm in low-level (at a frequency of 1KHz) via ROS. So it is not possible to do joint torque control in ROS

sinaaghli, 2023-01-11:
> I do not want to disapoint you @sinaaghli, but this example has been available for a long time now. It is linked in the second comment of this thread.
>
> However, this feature is not available [...]

is it an API requirement to run it in 1KHz? or more of an engineering opinion for a smoother control response?

HaeJunn, 2024-07-10:
Is torque control still impossible now?

martinleroux, 2024-07-10:
Hi @HaeJunn ,

As Felix said, torque control must be run at 1 kHz. This sort of communication rate is impossible in ROS1. Your only option therefore is to use the Low-level API outside of ROS. This is a limitation of ROS1, not a limitation of the arm.

Torque control via ROS2 is theoretically possible, but we have yet to implement the torque control driver. You would have the option to use the API to implement the ROS2 driver yourself should you want to.

I will be closing this issue since this will never be possible via ROS1.

closed 2024-07-10 by martinleroux