Kinovarobotics/ros_kortex #80: robotiq_2f_85 /gripper_cmd/feedback topic

stoudpav, 2020-02-26:
Dear colleagues,

the following topic does not seem to be working
_/my_gen3/robotiq_2f_85_gripper_controller/gripper_cmd/feedback_
is it just my problem or is it not implemented yet?

To get the gripper feedback I am currently using the **BaseCyclic_Feedback**, but I am not sure if that's the best solution. 

Thank you in advance and have a nice day
Pavel

alexvannobel, 2020-03-10:
Hi Pavel,

This topic is in fact the feedback of the Robotiq gripper action server. It publishes only when an action is ongoing in the action server for a gripper movement (typically when MoveIt is used to move the gripper). 

There are multiple ways to get the gripper feedback, and yes as you pointed out you can subscribe to `/my_gen3/base_feedback` and read the position value for motor #0 in the gripper feedback, but this is a bit tedious.

You can also subscribe to `/my_gen3/joint_states` and read the position value for the joint named "finger_joint". This message is more lightweight and doesn't contain other information that you might not need.

Don't hesitate to reopen the issue if you have more questions.
Best,
Alex

closed 2020-03-10 by alexvannobel

stoudpav, 2020-03-10:
Hi Alex,
thank you very much for your answer. 
I am trying to get as much information about the gripper as fast as I can.
My goal is to get enough feedback that I can use to get a "sense of touch" from the gripper.
So far I am using the **position** value in tandem with the **motor current** value, my problem is that the  `/my_gen3/base_feedback` is being send with a 40hz frequency if I am not mistaken and is kinda slow.

Is there something better and/or faster to get these feedback values?
I am okay with modifying or writing my own code if necessary . 
This is for my final thesis research, so thank you in advance.

Pavel