Kinovarobotics/ros_kortex #83: robotiq_2f_85 best way to get feedback?

stoudpav, 2020-03-10:
Hi Alex,
thank you very much for your answer. 
I am trying to get as much information about the gripper as fast as I can.
My goal is to get enough feedback that I can use to get a "sense of touch" from the gripper.
So far I am using the **position** value in tandem with the **motor current** value, my problem is that the  `/my_gen3/base_feedback` is being send with a 40hz frequency if I am not mistaken and is kinda slow.

Is there something better and/or faster to get these feedback values?
I am okay with modifying or writing my own code if necessary . 
This is for my final thesis research, so thank you in advance.

Pavel

_Originally posted by @stoudpav in https://github.com/Kinovarobotics/ros_kortex/issues/80#issuecomment-597324626_

alexvannobel, 2020-03-11:
Hi @stoudpav ,

It is true that as of release 2.2, the high level commands function only at 40Hz, however the feedback is on a different router and this one can work up to 1KHz. You won't be able to fetch feedback at 1KHz with a normal ROS1 installation though, as it's not really real-time friendly, but you can certainly get faster feedback than 40Hz. You can look at the `cyclic_data_publish_rate` parameter in the [kortex_driver.launch file](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/launch/kortex_driver.launch#L22). This is the parameter that controls the feedback loop in the driver (specifically [in this method](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/src/non-generated/driver/kortex_arm_driver.cpp#L495-L570)).

If that suits your use-case, I would advise you to start off by changing the argument in the launch file to have faster feedback. If you see it's not fast enough for you or that there is jitter in the publish, you could create your own MyGripperFeedback message (much smaller than joint_states or base_feedback) and publish it from this method. If you don't need the base_feedback or joint_states you could also comment them or reduce their publishing rate so your own topic has priority over those other topics.

Hope this helps,
Best,
Alex

alexvannobel, 2020-04-21:
Hi @stoudpav ,

Were you able to get faster feedback by changing this parameter?

Best,
Alex

stoudpav, 2020-04-21:
Hi @alexvannobel ,

Yes, I changed the launch file argument and it seems to be working. Thanks for your support and sorry for the delayed answer.

Best regards
Pavel

closed 2020-04-21 by stoudpav