Kinovarobotics/ros_kortex #84: catkin build functionality

ngeerlingsriwo, 2020-03-31:
I'm currently in a project where some packages require the use of 'catkin build' to compile. I followed the steps to install the ros_kortex packages, which compile with 'catkin_make' but fail with 'catkin build'. Is there some way that I could make it work with 'catkin build'?

I'm currently using ROS Melodic on Ubuntu 18. The error that I get when building e.g. kortex_driver is:
"
Could not find a package configuration file provided by "roscpp" with any of the following names:
roscppConfig.cmake
roscpp-config.cmake
"

polthergeist, 2020-03-31:
Hi @ngeerlingsriwo,
I will try to reproduce your problem here at my place but it may take some time. Because of that covid-19 thing, things have slowed down a bit. 
I'll keep you posted.

alexvannobel, 2020-04-21:
Hi @ngeerlingsriwo ,

I just opened a pull request to address this problem. I had the same behaviour as you, and I found out the conan_cmake_run call in kortex_driver's CMakeLists.txt affects the CMAKE_PREFIX_PATH and fails the builds with `catkin_make_isolated` and `catkin build`. I'll let you know when the PR is merged.

alexvannobel, 2020-04-21:
The PR has been merged to `kinetic-devel`.

I'll close the issue as the fix was pretty straightforward, but don't hesitate to reopen it if it persists.

Best regards,

Alex

closed 2020-04-21 by alexvannobel