Kinovarobotics/ros_kortex #85: Unnatural planned joint trajectories with moveit

huiwenzhang, 2020-04-13:
Hi, 
Basically, this is not an issue, but a question. I tried to control the Gen3 arm in Cartesian space with the MoveIt plugin. It seems that the planned joint trajectories are always very strange, which means the arm will flip over with big joint position changes even though the targets are very close in Cartesian space. I know the sample-based planning methods used in MoveIt can't guarantee smoothness  by its own, and the solutions are non-optimal, but it seems that the kinova arm encounters this problem more frequently, same problem happens when using the Jaco arm.  Why this problem happen? Is it because the hardware arm configurations? More importantly, are there any methods or tips which can output more natural planning motions?
Thanks,

polthergeist, 2020-04-13:
Hi @huiwenzhang ,
Thanks for communicating with us. Because of the covid-19 situation, we are working with less staff than usual but next week, more people are coming back to work and I'll get someone to answer your question. I'm sorry, I'm not very familiar with our moveit package. Still, I will take a few hours to check that out but don't expect a positive answer this week. Sorry about all that.

closed 2020-06-24 by huiwenzhang

juntuck, 2020-08-11:
Dear @huiwenzhang ,

Did you find out the reason for this? :)