Kinovarobotics/ros_kortex #89: Gripper parts not shown as connected in Gazebo

aamirhatim, 2020-04-30:
[image]
Here's my command:
`$ roslaunch kortex_gazebo spawn_kortex_robot.launch gripper:=robotiq_2f_85`
(See [output.txt](https://github.com/Kinovarobotics/ros_kortex/files/4561072/output.txt) for what gets printed to the terminal while everything starts up)

Everything starts as expected, but once the homing node unpauses Gazebo and the arm starts homing itself, the gripper parts start to disconnect and move away from the robot. Everything looks just fine Rviz though.

I'm assuming I'm doing something wrong in Gazebo or I'm missing some kind of package/plugin, but has this happened to anyone else? I've installed all package dependencies as well. I tried this on the 2f 140 and got the same behavior. This only happens when a gripper is attached - the arm itself seems fine. Apologies if this type of question should be posted elsewhere.

ROS Melodic
Ubuntu 18.04

phuicy, 2020-06-08:
I am also having this issue?

phuicy, 2020-06-08:
I have tried debugging the mimic plugin, and tuning the parameters, but to no luck.
I think there might be some self-collisions.

alexvannobel, 2020-06-08:
Hi @aamirhatim and @phuicy ,

It seems this issue slipped as it aired when we had less staff working because of COVID-19 measures. Really sorry about the delay.

The problem was indeed linked to the mimic joint plugin and I had to add the `<hasPID/> tag to stop the gripper pieces from flying around.

The PR is #98 , I'll let you know when it's merged. 

Again, sorry for the delay.
Best regards,

Alex

phuicy, 2020-06-09:
Ah you beat me to it. I was just here coming to comment that I have fixed it with the <hasPID/> tag.

thanks,
Guy

alexvannobel, 2020-06-09:
Hi,
I just merged the PR, I'll close the issue now. 
Best regards,
Alex

closed 2020-06-09 by alexvannobel