Kinovarobotics/ros_kortex #90: move gen3_lite using IK

aboulbarge, 2020-05-04:
Hello,
I try to make a simulation using IK, in a python code with the model of gen3_lite. I already installed the necessary packages, and I manage to launch the model of gen3_lite with moveit, but I don't know yet where I can find a code that allows me to manipulate this arm, using inverse kinematics with python.
Thanks.

alexvannobel, 2020-05-19:
Hi @aboulbarge ,

We don't have such an example yet. You can look into [this issue](https://github.com/Kinovarobotics/ros_kortex/issues/47) where a user wanted to use Twist commands in Gazebo.
You can also use MoveIt to move to Cartesian targets as in the [Python example](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_examples/src/move_it/example_move_it_trajectories.py#L205-L209).
I'll keep this issue open and update it if there is new development concerning IK with Gen3 lite.

Best regards,
Alex

alexvannobel, 2020-08-05:
Hi @aboulbarge ,

You can have a look at [the latest release](https://github.com/Kinovarobotics/ros_kortex/releases/tag/v2.2.2), which adds simulation improvements for Kortex arms.
Among the improvements, you can now use the `PlayCartesianTrajectory` ROS service to send Cartesian poses to the simulated arm. The IK is calculated with KDL in the `kortex_driver`. 

As this is now possible in simulation, I'll close this issue. Don't hesitate to reopen it if you encounter problems, or open a new issue.

Best,
Alex

closed 2020-08-05 by alexvannobel