Kinovarobotics/ros_kortex #91: Catkin_make kortex_driver on ARM hardware

ngeerlingsriwo, 2020-05-15:
For deployment I would like to run the ROS Kortex packages on an ARM instead of an x86 device. I have created a Dockerfile (inheriting from ros:melodic-robot) which performs the installation for Conan and the packages according to your manual. When ```docker build``` reaches the ```catkin_make_isolated``` step, the kortex_driver package cannot be built. I fear that this has to do with the ARM core, according to the error message (shown below). I have tried to build the exact same Docker file on my laptop (x86), and there the build is successful and I can run ROS within Docker. 

Is ARM hardware supported? If so, how should I proceed?

My Dockerfile:
```
# Use the official image as a parent image.
FROM ros:melodic-robot
    
RUN rm -rf /var/lib/apt/lists/
    
# Set up Conan for Kinova Kortex
RUN apt-get update \
    && apt-get install -y python3 python3-pip \
    && python3 -m pip install conan \
    && conan config set general.revisions_enabled=1 \
    && conan profile new default --detect > /dev/null \
    && conan profile update settings.compiler.libcxx=libstdc++11 default

# Acquire ROS dependencies
RUN mkdir -p catkin_ws/src \
    && cd catkin_ws/src \
    && git clone https://github.com/Kinovarobotics/ros_kortex.git \
    && cd ../ \
    && rosdep install --from-paths src --ignore-src -y

RUN /bin/bash -c '. /opt/ros/melodic/setup.bash; cd catkin_ws/; catkin_make_isolated; source devel_isolated/setup.bash'
```

The error message:
```
==> Processing catkin package: 'kortex_driver'
==> Creating build directory: 'build_isolated/kortex_driver'
==> Building with env: '/catkin_ws/devel_isolated/gen3_robotiq_2f_85_move_it_config/env.sh'
==> cmake /catkin_ws/src/ros_kortex/kortex_driver -DCATKIN_DEVEL_PREFIX=/catkin_ws/devel_isolated/kortex_driver -DCMAKE_INSTALL_PREFIX=/catkin_ws/install_isolated -G Unix Makefiles in '/catkin_ws/build_isolated/kortex_driver'
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /catkin_ws/devel_isolated/kortex_driver
-- Using CMAKE_PREFIX_PATH: /catkin_ws/devel_isolated/gen3_robotiq_2f_85_move_it_config;/catkin_ws/devel_isolated/gen3_robotiq_2f_140_move_it_config;/catkin_ws/devel_isolated/gen3_move_it_config;/catkin_ws/devel_isolated/gen3_lite_gen3_lite_2f_move_it_config;/catkin_ws/devel_isolated/kortex_description;/catkin_ws/devel_isolated/kortex_control;/catkin_ws/devel_isolated/gazebo_grasp_plugin;/catkin_ws/devel_isolated/gazebo_version_helpers;/catkin_ws/devel_isolated/ddynamic_reconfigure;/catkin_ws/devel_isolated/dynamic_reconfigure;/catkin_ws/devel_isolated/armadillo_matrix;/catkin_ws/devel_isolated/aap_msgs;/opt/ros/melodic
-- This workspace overlays: /catkin_ws/devel_isolated/gen3_robotiq_2f_85_move_it_config;/catkin_ws/devel_isolated/gen3_robotiq_2f_140_move_it_config;/catkin_ws/devel_isolated/gen3_move_it_config;/catkin_ws/devel_isolated/gen3_lite_gen3_lite_2f_move_it_config;/catkin_ws/devel_isolated/kortex_description;/catkin_ws/devel_isolated/kortex_control;/catkin_ws/devel_isolated/gazebo_grasp_plugin;/catkin_ws/devel_isolated/gazebo_version_helpers;/catkin_ws/devel_isolated/ddynamic_reconfigure;/catkin_ws/devel_isolated/dynamic_reconfigure;/catkin_ws/devel_isolated/armadillo_matrix;/catkin_ws/devel_isolated/aap_msgs;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /catkin_ws/build_isolated/kortex_driver/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.23
-- BUILD_SHARED_LIBS is on
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
-- kortex_driver: 408 messages, 254 services
-- Conan: checking conan executable in path
-- Conan: Found program /usr/local/bin/conan
-- Conan: Version found Conan version 1.25.1

-- Conan: Adding kinova_public remote repository (https://artifactory.kinovaapps.com/artifactory/api/conan/conan-public)
WARN: Remotes registry file missing, creating default one in /root/.conan/remotes.json
-- Conan: Automatic detection of conan settings from cmake
-- Conan: Settings= -s;build_type=Release;-s;compiler=gcc;-s;compiler.version=7;-s;compiler.libcxx=libstdc++11;-s;kortex_api_cpp:compiler=gcc;-s;kortex_api_cpp:compiler.version=5;-s;kortex_api_cpp:compiler.libcxx=libstdc++11
-- Conan executing: conan install /catkin_ws/src/ros_kortex/kortex_driver/conanfile.py -s build_type=Release -s compiler=gcc -s compiler.version=7 -s compiler.libcxx=libstdc++11 -s kortex_api_cpp:compiler=gcc -s kortex_api_cpp:compiler.version=5 -s kortex_api_cpp:compiler.libcxx=libstdc++11 -g=cmake --update
Configuration:
[settings]
arch=armv8
arch_build=armv8
build_type=Release
compiler=gcc
compiler.libcxx=libstdc++11
compiler.version=7
os=Linux
os_build=Linux
kortex_api_cpp:compiler=gcc
kortex_api_cpp:compiler.version=5
kortex_api_cpp:compiler.libcxx=libstdc++11
[options]
[build_requires]
[env]

kortex_api_cpp/2.2.0-r.31@kortex/stable: Not found in local cache, looking in remotes...
kortex_api_cpp/2.2.0-r.31@kortex/stable: Trying with 'conan-center'...
kortex_api_cpp/2.2.0-r.31@kortex/stable: Trying with 'kinova_public'...
Downloading conanmanifest.txt
Downloading conanfile.py
kortex_api_cpp/2.2.0-r.31@kortex/stable: Downloaded recipe revision 571177d781b9c030189c0b4d126b212dd16d9e7d
conanfile.py: Installing package
Requirements
    kortex_api_cpp/2.2.0-r.31@kortex/stable from 'kinova_public' - Downloaded
Packages
    kortex_api_cpp/2.2.0-r.31@kortex/stable:624ada0747b75761a27a9e18e13c874a052e76b0 - Missing

Installing (downloading, building) binaries...

ERROR: Missing binary: kortex_api_cpp/2.2.0-r.31@kortex/stable:624ada0747b75761a27a9e18e13c874a052e76b0
kortex_api_cpp/2.2.0-r.31@kortex/stable: WARN: Can't find a 'kortex_api_cpp/2.2.0-r.31@kortex/stable' package for the specified settings, options and dependencies:
- Settings: arch=armv8, build_type=Release, compiler=gcc, compiler.libcxx=libstdc++11, compiler.version=5, os=Linux
- Options: shared=False, target=x86
- Dependencies: 
- Requirements: 
- Package ID: 624ada0747b75761a27a9e18e13c874a052e76b0

ERROR: Missing prebuilt package for 'kortex_api_cpp/2.2.0-r.31@kortex/stable'
Try to build it from sources with "--build kortex_api_cpp"
Or read "http://docs.conan.io/en/latest/faq/troubleshooting.html#error-missing-prebuilt-package"

CMake Error at cmake/conan.cmake:399 (message):
  Conan install failed='1'
Call Stack (most recent call first):
  cmake/conan.cmake:485 (conan_cmake_install)
  CMakeLists.txt:64 (conan_cmake_run)

-- Configuring incomplete, errors occurred!
See also "/catkin_ws/build_isolated/kortex_driver/CMakeFiles/CMakeOutput.log".
See also "/catkin_ws/build_isolated/kortex_driver/CMakeFiles/CMakeError.log".
<== Failed to process package 'kortex_driver': 
  Command '['/catkin_ws/devel_isolated/gen3_robotiq_2f_85_move_it_config/env.sh', 'cmake', '/catkin_ws/src/ros_kortex/kortex_driver', '-DCATKIN_DEVEL_PREFIX=/catkin_ws/devel_isolated/kortex_driver', '-DCMAKE_INSTALL_PREFIX=/catkin_ws/install_isolated', '-G', 'Unix Makefiles']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /catkin_ws/build_isolated/kortex_driver && /catkin_ws/devel_isolated/gen3_robotiq_2f_85_move_it_config/env.sh cmake /catkin_ws/src/ros_kortex/kortex_driver -DCATKIN_DEVEL_PREFIX=/catkin_ws/devel_isolated/kortex_driver -DCMAKE_INSTALL_PREFIX=/catkin_ws/install_isolated -G 'Unix Makefiles'
```

alexvannobel, 2020-05-19:
Hi @ngeerlingsriwo ,

We do build the Kortex API for a couple ARM platforms, but the default "target" option in our Conan repository is "x86".
To change it, you will have to change the call to [conan_cmake_run](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/CMakeLists.txt#L64-L68) in the CMakeLists.txt file.

We build for three ARMv8 platforms (even though the conan manifest says armv7, that's something we need to change on our side), so you could try them all to see which is compatible with your platform during linking of ROS Kortex : 

- For Artik710 platform : 
```
  conan_cmake_run(CONANFILE conanfile.py
                  UPDATE
                  BASIC_SETUP CMAKE_TARGETS
                  NO_OUTPUT_DIRS
                  SETTINGS kortex_api_cpp:arch=armv7 kortex_api_cpp:compiler=gcc kortex_api_cpp:compiler.version=5 kortex_api_cpp:compiler.libcxx=libstdc++11
                  OPTIONS kortex_api_cpp:target=artik710)
```
- For IMX6 platform : 
```
  conan_cmake_run(CONANFILE conanfile.py
                  UPDATE
                  BASIC_SETUP CMAKE_TARGETS
                  NO_OUTPUT_DIRS
                  SETTINGS kortex_api_cpp:arch=armv7 kortex_api_cpp:compiler=gcc kortex_api_cpp:compiler.version=6.4 kortex_api_cpp:compiler.libcxx=libstdc++11
                  OPTIONS kortex_api_cpp:target=imx6)
```
- For NVidia Jetson Xavier : 
```
  conan_cmake_run(CONANFILE conanfile.py
                  UPDATE
                  BASIC_SETUP CMAKE_TARGETS
                  NO_OUTPUT_DIRS
                  SETTINGS kortex_api_cpp:arch=armv7 kortex_api_cpp:compiler=gcc kortex_api_cpp:compiler.version=7 kortex_api_cpp:compiler.libcxx=libstdc++11
                  OPTIONS kortex_api_cpp:target=jetson)
```
I'll leave this issue open to make sure we add this to the README file, or probably in fact just make and document a CMake argument to build for arm64.

Hope this helps,
Best regards,
Alex

ngeerlingsriwo, 2020-05-19:
Thank you, I will try it that way

alexvannobel, 2020-05-20:
Hi @ngeerlingsriwo ,
I merged the instructions to build the ROS Kortex packages for ARMv8. 
Did it work for you?

ngeerlingsriwo, 2020-05-25:
Yes I can build ros_driver now!

closed 2020-05-25 by alexvannobel