Kinovarobotics/ros_kortex #92: Gen3Lite2f gripper actuation

ngeerlingsriwo, 2020-05-19:
When the Gen3Lite robot is controlled directly by the Xbox controller, I am able to make jog movements with the 2f gripper. When I look at the ROS topics which I can actuate the gripper with, I can only find the GripperCommand action, which is position control. 

Is there a way to do velocity control / jogging from ROS without having to use the position control action?

If so, is it also possible to do force control (based on the current, as I could not find force feedback for the gripper)? I would like to pick up soft items without squeezing them, so I would like to send a desired force and a maximum position displacement to the gripper.

alexvannobel, 2020-05-20:
Hi @ngeerlingsriwo ,

**About speed control :**
Yes you can control the gripper in velocity from ROS. There is a [C++ example](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/106-Gripper_command/01-gripper_command.cpp#L115-L160) in the Kortex repository which does that, so you can refer to that if you want more details.

In essence, you will have to call the SendGripperCommand RPC. I call it with `rosservice` as such:
```
rosservice call /my_gen3_lite/base/send_gripper_command "input:
  mode: 2
  gripper:
    finger:
    - finger_identifier: 0
      value: 0.1
  duration: 0"
```
You can find the definition of the RPC [here](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/srv/generated/base/SendGripperCommand.srv), and the definition for the `mode` enum [here](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/msg/generated/base/GripperMode.msg).
You can replicate that in rospy or roscpp if you need to.

**About force/current control**
Although the enum value exists in GripperMode.msg, force/current control with the Gen3 lite gripper is not supported for now. 

Tell me if you manage to use the velocity control mode!
Hope this helps,
Best regards,
Alex

ngeerlingsriwo, 2020-05-25:
This gave a lot of insights, I've got it working now. Thank you!

alexvannobel, 2020-05-25:
Glad ir worked!

closed 2020-05-25 by alexvannobel