Kinovarobotics/ros_kortex #96: Conan returns "cannot import name 'select_autoescape'"

flymaxty, 2020-06-01:
Hi,
I'm trying to use conan to download api, but conan returns the following error when exec `conan config set general.revisions_enabled=1`

```bash
ysc@ysc-NUC8i7BEH:~/kinova$ conan config set general.revisions_enabled=1
Traceback (most recent call last):
  File "/usr/local/bin/conan", line 5, in <module>
    from conans.conan import run
  File "/usr/local/lib/python3.5/dist-packages/conans/conan.py", line 3, in <module>
    from conans.client.command import main
  File "/usr/local/lib/python3.5/dist-packages/conans/client/command.py", line 16, in <module>
    from conans.client.conan_api import Conan, default_manifest_folder, _make_abs_path, ProfileData
  File "/usr/local/lib/python3.5/dist-packages/conans/client/conan_api.py", line 11, in <module>
    from conans.client.cache.cache import ClientCache
  File "/usr/local/lib/python3.5/dist-packages/conans/client/cache/cache.py", line 7, in <module>
    from jinja2 import Environment, select_autoescape
ImportError: cannot import name 'select_autoescape'
```

I'm using Ubuntu 18.04 & ROS Melodic

alexvannobel, 2020-06-01:
Hi @flymaxty ,

That's a weird one, it seems related to `jinja2`.
Can you try reinstalling it and conan too with `python3 -m pip install --upgrade jinja2 conan`?
If this persists, can you tell me what jinja2 and conan version were installed?

Hope this helps,
Best,
Alex

flymaxty, 2020-06-08:
> Hi @flymaxty ,
> 
> That's a weird one, it seems related to `jinja2`.
> Can you try reinstalling it and conan too with `python3 -m pip install --upgrade jinja2 conan`?
> If this persists, can you tell me what jinja2 and conan version were installed?
> 
> Hope this helps,
> Best,
> Alex

Hi Alex,
Sorry for the late reply, reinstall conan ad upgrade jinjia2 solve the problem.
Thanks for your help! :)

closed 2020-06-08 by flymaxty