Kinovarobotics/ros_kortex #106: Sending gripper command with ExecuteAction service fails

ohyeahthatguy81, 2020-07-06:
I'm trying to open/close our Robotiq 2F 140 gripper using the ExecuteAction service. I've used this service for sending poses (ConstrainedPose) and joint values (ConstrainedJointAngles) with success. However, once I've called the service and wait for the gripper command to be completed, my action notification handler always reports an aborted action (action_event == ActionEvent::ACTION_ABORT). It never matters which value I give (0.0, 0.5, 1.0 as examples).

This is what my method looks like:

```
bool ArmControl::manipulateGripper(const double& iGripperPosition, bool iWait)
{
    bool gripperManipulated = false;

    kortex_driver::GripperCommand gripperCommand;
    
    kortex_driver::Finger finger;
    /* 0 as we only have one "finger" to manipulate */
    finger.finger_identifier = 0;
    finger.value = iGripperPosition;

    gripperCommand.mode = kortex_driver::GripperMode::GRIPPER_POSITION;
    gripperCommand.gripper.finger.push_back(finger);
	
	ros::ServiceClient executeActionClient = m_nodeHandle.serviceClient<kortex_driver::ExecuteAction>("/" + m_robotName + "/base/execute_action");
	kortex_driver::ExecuteAction executeActionObj;
	
	executeActionObj.request.input.oneof_action_parameters.send_gripper_command.push_back(gripperCommand);
	executeActionObj.request.input.name = "GripperCommand";
	executeActionObj.request.input.handle.identifier = m_eventID;

	ROS_DEBUG_STREAM(executeActionObj.request.input);

	gripperManipulated = executeActionClient.call(executeActionObj);

	if (gripperManipulated)
	{
		ROS_DEBUG_STREAM("Gripper command was sent to the robot. Waiting on robot..");

		while (!actionCompleted && !actionAborted)
		{
			ros::spinOnce();
		}

		if (actionCompleted)
		{
			ROS_DEBUG_STREAM("Gripper command was sent.");
		}

		if (actionAborted)
		{
			ROS_ERROR_STREAM("Gripper command could not be sent!");
			gripperManipulated = false;
		}

		actionCompleted = actionAborted = false;
		m_eventID++;
	}
	else
	{
		ROS_ERROR_STREAM("Failed to send gripper command to the robot!");
	}      

    return gripperManipulated;
}
```

My notification callback:

```
void ArmControl::actionNotificationHandler(const kortex_driver::ActionNotification::ConstPtr& obj)
{
    ROS_DEBUG_STREAM("actionNotificationHandler:\n" << (*obj));
    switch (obj->action_event)
    {
        case kortex_driver::ActionEvent::ACTION_END:
        {
            if (obj->handle.identifier == m_eventID)
            {
                actionCompleted = true;
            }            
            
            ROS_DEBUG_STREAM("Action ended.");
        }
        break;

        case kortex_driver::ActionEvent::ACTION_ABORT:
        {
            if (obj->handle.identifier == m_eventID)
            {
                actionAborted = true;
            }

            ROS_DEBUG_STREAM("Action aborted!");
        }
        break;        
    }
}
```

I've also noticed when calling `rosservice echo /my_gen3/action_topic`, a message is never published when opening/closing the gripper using the X-Box controller - only when moving the robot. The robot is not moving when the manipulateGripper(....) is being called, so I'm little puzzled why the action is prematurely aborted.

My current workaround is just to call the send_gripper_command service and wait for some fixed amount of time (such as 1 second), but I would rather prefer to use ExecuteAction service and let it notify me.

alexvannobel, 2020-07-15:
Hi @ohyeahthatguy81 ,

Thanks for providing the code! I reproduced your issue quickly and tried a couple things, and the fix is : put `finger.finger_identifier = 1;`. 
I don't have a definitive explanation as to why 1 instead of 0, but I'm pretty sure it's because Protobuf messages's default `int` value is 0, and so the firmware probably can't make the difference between specifying finger #0 and not specifying a finger.

Tell me how this goes on your side and don't hesitate to reopen the issue or create a new one.
Best,
Alex

closed 2020-07-15 by alexvannobel

ohyeahthatguy81, 2020-07-16:
> Hi @ohyeahthatguy81 ,
>
> Thanks for providing the code! I reproduced your issue quickly and tried a couple things, and the fix is : put `finger.finger_identifier = 1;`.
> I don't have a definitive [...]

Hello @alexvannobel,

I can confirm that the workaround works, thank you!