Kinovarobotics/ros_kortex #113: Error when calling catkin_make (with potential fix)

aamirhatim, 2020-08-06:
Ubuntu 16.04
ROS Kinetic

I'm getting the following errors when running `catkin_make`:

```
/home/administrator/catkin_ws/src/ros_kortex/kortex_examples/src/full_arm/example_cartesian_poses_with_notifications.cpp:36:18: error: variable 'std::atomic<int> last_action_notification_event' has initializer but incomplete type
 std::atomic<int> last_action_notification_event{0};
                  ^
/home/administrator/catkin_ws/src/ros_kortex/kortex_examples/src/full_arm/example_cartesian_poses_with_notifications.cpp:37:18: error: variable 'std::atomic<int> last_action_notification_id' has initializer but incomplete type
 std::atomic<int> last_action_notification_id{0};
                  ^
make[2]: *** [ros_kortex/kortex_examples/CMakeFiles/example_cartesian_poses_with_notifications_cpp.dir/src/full_arm/example_cartesian_poses_with_notifications.cpp.o] Error 1
make[1]: *** [ros_kortex/kortex_examples/CMakeFiles/example_cartesian_poses_with_notifications_cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /home/administrator/catkin_ws/src/ros_kortex/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp:13:0:
/home/administrator/catkin_ws/src/ros_kortex/kortex_driver/include/kortex_driver/non-generated/kortex_arm_simulation.h:172:23: error: field 'm_is_action_being_executed' has incomplete type 'std::atomic<bool>'
     std::atomic<bool> m_is_action_being_executed;
                       ^
In file included from /usr/include/c++/5/bits/shared_ptr_atomic.h:33:0,
                 from /usr/include/c++/5/memory:83,
                 from /usr/include/boost/config/no_tr1/memory.hpp:21,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:23,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/boost/format/alt_sstream.hpp:21,
                 from /usr/include/boost/format/internals.hpp:23,
                 from /usr/include/boost/format.hpp:38,
                 from /usr/include/boost/math/policies/error_handling.hpp:31,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/kinetic/include/ros/time.h:58,
                 from /opt/ros/kinetic/include/ros/ros.h:38,
                 from /home/administrator/catkin_ws/src/ros_kortex/kortex_driver/include/kortex_driver/non-generated/kortex_arm_simulation.h:17,
                 from /home/administrator/catkin_ws/src/ros_kortex/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp:13:
/usr/include/c++/5/bits/atomic_base.h:126:12: note: declaration of 'struct std::atomic<bool>'
     struct atomic;
            ^
In file included from /home/administrator/catkin_ws/src/ros_kortex/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp:13:0:
/home/administrator/catkin_ws/src/ros_kortex/kortex_driver/include/kortex_driver/non-generated/kortex_arm_simulation.h:173:23: error: field 'm_action_preempted' has incomplete type 'std::atomic<bool>'
     std::atomic<bool> m_action_preempted;
                       ^
In file included from /usr/include/c++/5/bits/shared_ptr_atomic.h:33:0,
                 from /usr/include/c++/5/memory:83,
                 from /usr/include/boost/config/no_tr1/memory.hpp:21,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:23,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/boost/format/alt_sstream.hpp:21,
                 from /usr/include/boost/format/internals.hpp:23,
                 from /usr/include/boost/format.hpp:38,
                 from /usr/include/boost/math/policies/error_handling.hpp:31,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/kinetic/include/ros/time.h:58,
                 from /opt/ros/kinetic/include/ros/ros.h:38,
                 from /home/administrator/catkin_ws/src/ros_kortex/kortex_driver/include/kortex_driver/non-generated/kortex_arm_simulation.h:17,
                 from /home/administrator/catkin_ws/src/ros_kortex/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp:13:
/usr/include/c++/5/bits/atomic_base.h:126:12: note: declaration of 'struct std::atomic<bool>'
     struct atomic;
            ^
In file included from /home/administrator/catkin_ws/src/ros_kortex/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp:13:0:
/home/administrator/catkin_ws/src/ros_kortex/kortex_driver/include/kortex_driver/non-generated/kortex_arm_simulation.h:174:22: error: field 'm_current_action_type' has incomplete type 'std::atomic<int>'
     std::atomic<int> m_current_action_type;
                      ^
In file included from /usr/include/c++/5/bits/shared_ptr_atomic.h:33:0,
                 from /usr/include/c++/5/memory:83,
                 from /usr/include/boost/config/no_tr1/memory.hpp:21,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:23,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/boost/format/alt_sstream.hpp:21,
                 from /usr/include/boost/format/internals.hpp:23,
                 from /usr/include/boost/format.hpp:38,
                 from /usr/include/boost/math/policies/error_handling.hpp:31,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/kinetic/include/ros/time.h:58,
                 from /opt/ros/kinetic/include/ros/ros.h:38,
                 from /home/administrator/catkin_ws/src/ros_kortex/kortex_driver/include/kortex_driver/non-generated/kortex_arm_simulation.h:17,
                 from /home/administrator/catkin_ws/src/ros_kortex/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp:13:
/usr/include/c++/5/bits/atomic_base.h:126:12: note: declaration of 'struct std::atomic<int>'
     struct atomic;
            ^
make[2]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_arm_driver_implementation.dir/src/non-generated/driver/kortex_arm_simulation.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_arm_driver_implementation.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```

I've also attached the rest of the output of `catkin_make` as well in case you need it. 

I added `#include <atomic>` to the following files fixed the problem:
- `ros_kortex/kortex_examples/src/full_arm/example_cartesian_poses_with_notifications.cpp`
- `ros_kortex/kortex_driver/include/kortex_driver/non-generated/kortex_arm_simulation.h`

Not sure if that is where these includes should go but I figured you would know that better than me!

[ros_kortex_catkin_make.txt](https://github.com/Kinovarobotics/ros_kortex/files/5037940/ros_kortex_catkin_make.txt)

alexvannobel, 2020-08-07:
Hi @aamirhatim ,

I just created a PR for this issue. Merging it will close the issue.
Out of curiosity, what compiler do you have? I did not have this problem with gcc 7.5 (ROS Melodic).

Best,
Alex

aamirhatim, 2020-08-07:
I have 5.4 so that may be why. I'm currently building this package on a Clearpath Ridgeback robot

PR #115 (2020-08-07): Atomic compilation problem and put back --inorder
Fixes #113 and fixes #114
closed 2020-08-07 by alexvannobel