Kinovarobotics/ros_kortex #114: Explicitly implement --inorder processing

aamirhatim, 2020-08-06:
Ubuntu 16.04
ROS Kinetic

I ran into this error when launching `kinova_driver.launch`:
```
xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
variable filename is supported with --inorder option only None None
when processing file: /home/administrator/catkin_ws/src/ros_kortex/kortex_description/robots/kortex_robot.xacro
included from: /home/administrator/catkin_ws/src/ros_kortex/kortex_description/robots/gen3.xacro
```
Adding the `--inorder` option inside the description load param tag fixed the problem since the load command does indeed rely on variables within the filename. Not sure how many files this actually impacts but the fix itself is small.

alexvannobel, 2020-08-07:
Hi @aamirhatim ,

I just created a PR for this issue. Merging it will close the issue.
I tested the release on ROS Melodic, for which I got this warning instead of what you had : 
`xacro: in-order processing became default in ROS Melodic. You can drop the option.`
I thought it was also default in Kinetic, but it seems I was wrong!

Thanks for the feedback,
Best,
Alex

aamirhatim, 2020-08-07:
Awesome, thanks Alex. I originally built on Melodic and didn't see the issue either. Only noticed it once I was building on my actual robot! (Clearpath Ridgeback running 16.04/Kinetic)

closed 2020-08-07 by alexvannobel

pchang17, 2021-05-29:
Ubuntu 16.04
ROS Kinetic

I ran into the same issue when I did `roslaunch gen3_robotiq_2f_85_move_it_config demo.launch`. I fixed the issue by adding `--inorder` at line 18 in `planning_context.launch`:
`<param if="$(arg load_robot_description)" name="$(arg robot_description)" command="xacro --inorder arm:=gen3 gripper:=robotiq_2f_85 '$(find kortex_description)/robots/gen3_robotiq_2f_85.xacro'"/>`

PR #115 (2020-08-07): Atomic compilation problem and put back --inorder
Fixes #113 and fixes #114