Kinovarobotics/ros_kortex #116: action_topic notification event ends with ACTION_END even though the pose was not reached

ohyeahthatguy81, 2020-08-14:
Hello,

I subscribe to the /.../action_topic topic so that I can safely wait until a target pose has been reached. To accomplish this task, I call the /.../base/execute_action service to reach a pose (using oneof_action_parameters.reach_pose). 

The pose I'm trying to reach is definitely reachable, and the target pose is a Z offset of the robot's current position. I'm not specifying any constraints, ergo the robot's set soft limits are taken into effect. I'm using a Gen3 robot, if that helps.

What I've observed so far:
a) If I set a Cartesian trajectory speed of 0.2 m/s and an angular trajectory speed of 30 °/s for each joint, the target pose is never reached, **but the action_topic callback ends with ACTION_END**. In fact, the higher the Cartesian trajectory speed (such as 0.5 m/s, the further the distance the robot stops away from the target pose!
b) If I set a Cartesian trajectory speed of 0.2 m/s and an angular trajectory speed of, lets say, 50 °/s for each joint, the target pose is reached without issue.

I can visually confirm that with observation (a) that the target pose is never reached. I can also confirm this by comparing the target pose with the robot's pose after the action_topic event handler has completed.

The point I'm trying to make:
If the robot never reaches the target pose - irrespective of set soft limits - I would expect that the action_topic event handler to end with ACTION_ABORT. The action_topic event handler is reliable, but only if a "magical" Cartesian and angular trajectory speed combination has been set.  I've also noticed that sometimes the robot starts moving to a position, only then saying that it cannot reach said position after traveling some distance.

Some questions:
1. When moving the robot with a Cartesian command (pose), what set speed is actually taken into account?
2. Why does the robot sometimes move towards a set pose, only to find out after traveling some set distance that it cannot reach it?

alexvannobel, 2020-11-23:
Hi @ohyeahthatguy81 ,
Sorry for the long delay.

For your observations a) and b) I'll ask the robotics control team what is the exact behaviour and I'll come back to you. It indeed seems unintended that if the robot does not reach the goal it sends an `ACTION_END` instead of an `ACTION_ABORT`.

For your questions:
1. If you set a Cartesian linear speed with the trajectory, this will be the maximum linear speed permitted during the trajectory (if the limit is lower than the Cartesian twist soft limit). If you don't put any limit or put a limit higher than the soft limit, the soft limit is used as maximum velocity. You can configure the robot's speed limits in the Configuration->Speed Limits tab in the WebApp.
2. There is no offline validation of a Cartesian trajectory before playing it, so the calculation is done online when it's playing (calculating the next increment to reach the goal every millisecond). If the arm runs in a singularity or outside its workspace, it will trigger an error and not be able to reach the goal pose.

Hope that answers,
Best,
Alex

felixmaisonneuve, 2021-07-26:
Hi @ohyeahthatguy81,

Do you still have an issue?
I will wait for a couple weeks and this issue if I don't get any response.

Regards,
Felix

closed 2021-08-20 by felixmaisonneuve