Kinovarobotics/ros_kortex #119: Colliding namespaces in ROS internal condition_variable.h

Dereen, 2020-08-19:
Hello, 
Im unable to compile your code. I followed your instructions and it always fails at kortex_driver. I tried create my own package where I included the same problematic header `/opt/ros/melodic/include/ros/internal/condition_variable.h` and the problem disappeared. 
Result from `catkin_make &> build.log` is in attached file.
[build.log](https://github.com/Kinovarobotics/ros_kortex/files/5095112/build.log)

Thank you for your assistance.
Valentýn Číhala

alexvannobel, 2020-08-27:
Hi @Dereen ,

I am not able to reproduce this compilation issue on my end, on ROS Melodic with GCC 7.5. 

What compiler version do you have and in what file have you included this header?

Thanks,
Alex

deepparikh, 2020-08-31:
I am also facing the same build issue, I have attached the build log with this post.
OS - Ubuntu 18.04
ROS Melodic 
GCC 7.5
CMake 3.18.1
[build_log.txt](https://github.com/Kinovarobotics/ros_kortex/files/5148514/build_log.txt)

juntuck, 2020-09-08:
Same issue here...

Ubuntu 18.04
ROS Melodic
GCC 7.5.0
CMake 3.10.2

[build_log.txt](https://github.com/Kinovarobotics/ros_kortex/files/5187170/build_log.txt)

felixmaisonneuve, 2021-06-16:
Hi everyone,

Is anybody still having this issue?

Best,
Felix

felixmaisonneuve, 2021-07-26:
Since nobody responded, I will close this issue. 

@Dereen, @deepparikh and @juntuck, if you still have problems, open a new issue or comment/re-open this issue and I will help you.

closed 2021-07-26 by felixmaisonneuve