Kinovarobotics/ros_kortex #122: compiler not defined for compiler.libcxx

RaduCorcodel, 2020-09-04:
System: Ubuntu 16.04, fresh install of ROS-kinetic

Tried to install ros_kortex following the instructions and got this error message:

```-- Boost version: 1.58.0
-- +++ processing catkin package: 'kortex_driver'
-- ==> add_subdirectory(ros_kortex/kortex_driver)
CONAN_TARGET_PLATFORM is x86
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
-- kortex_driver: 408 messages, 254 services
-- Conan: checking conan executable in path
-- Conan: Found program /usr/local/bin/conan
-- Conan: Version found Conan version 1.23.0

-- Conan: Adding kinova_public remote repository (https://artifactory.kinovaapps.com/artifactory/api/conan/conan-public)
-- Conan: Automatic detection of conan settings from cmake
-- Conan: Settings= -s;build_type=Release;-s;compiler=gcc;-s;compiler.version=5;-s;compiler.libcxx=libstdc++11;-s;kortex_api_cpp:compiler=gcc;-s;kortex_api_cpp:compiler.version=5;-s;kortex_api_cpp:compiler.libcxx=libstdc++11
-- Conan executing: conan install /homes/user/kinova_workspace/src/ros_kortex/kortex_driver/conanfile.py -s build_type=Release -s compiler=gcc -s compiler.version=5 -s compiler.libcxx=libstdc++11 -s kortex_api_cpp:compiler=gcc -s kortex_api_cpp:compiler.version=5 -s kortex_api_cpp:compiler.libcxx=libstdc++11 -g=cmake --update
ERROR: compiler not defined for compiler.libcxx
Please define compiler value first too
CMake Error at ros_kortex/kortex_driver/cmake/conan.cmake:399 (message):
  Conan install failed='1'
Call Stack (most recent call first):
  ros_kortex/kortex_driver/cmake/conan.cmake:485 (conan_cmake_install)
  ros_kortex/kortex_driver/CMakeLists.txt:70 (conan_cmake_run)

-- Configuring incomplete, errors occurred!
See also "/homes/user/kinova_workspace/build/CMakeFiles/CMakeOutput.log".
See also "/homes/user/kinova_workspace/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed```

alexvannobel, 2020-11-23:
Hi @RaduCorcodel ,
Sorry for the long delay.

This is a weird error message. In [this issue](https://github.com/conan-io/conan/issues/615) it seems like an order problem but I've never seen that.
What's the output of `conan profile show default` in a terminal?

tsoud, 2021-04-20:
Hi, 
I'm running into this exact same issue. The output from `CMake` is identical to that of the original poster.
`ERROR: compiler not defined for compiler.libcxx
Please define compiler value first too
CMake Error at ros_kortex/kortex_driver/cmake/conan.cmake:399 (message):
  Conan install failed='1'`

This is the output from `conan profile show default`:
`[settings]
os=Linux
os_build=Linux
arch=x86_64
arch_build=x86_64
build_type=Release
compiler.libcxx=libstdc++11
[options]
[build_requires]
[env]
`

After reviewing the issue referenced above, it doesn't seem that ordering (in the `install` command) is the problem: 
`-- Conan executing: conan install /home/user/directory/subdir/downloaded_robots/kinova_gen3/ros_ws/src/ros_kortex/kortex_driver/conanfile.py -s build_type=Release -s compiler=gcc -s compiler.version=9 -s compiler.libcxx=libstdc++11 -s kortex_api_cpp:compiler=gcc -s kortex_api_cpp:compiler.version=5 -s kortex_api_cpp:compiler.libcxx=libstdc++11 -g=cmake --update` where `user/directory/subdir` is the parent directory. 

Any recommendations?
Thanks

polthergeist, 2021-04-22:
Hi @tsoud and @RaduCorcodel ,
Could you try to install manually your conan profile using this command:
`conan install . --profile /abs/path/to/profile`
You can put your profile in a .cfg file that should look like this:
[image]
 You can modify those settings to match the one you need.

Keep me posted,

tsoud, 2021-04-24:
Hi @polthergeist,
Thanks for the advice. Unfortunately, I followed the recommendations and still got the same error. I'm sure I'm missing something but not sure where to look. 

Further info:
I added a robotics.cfg with the exact parameters you provided to my ~/.conan/profiles directory (this is also where the default config is). I ran `conan . --profile ~/.conan/profiles/robotics.cfg` from the directory suggested by the conan helpfile:
[image]

In this case, the directory ('.') containing `conanfile.py` is: /many/parent/dirs/src/ros_ws/src/ros_kortex/kortex_driver/ if I understood the instructions correctly.

Another important detail is that I'm running all this from within a conda environment. I don't know if this affects anything but it seems everything is set up correctly in the environment. Any other ideas welcome. Thanks in advance.

felixmaisonneuve, 2021-06-18:
Hi @tsoud,

I see your are using ROS Neotic. We do not officially support it yet, but I will be working on ROS Melodic -> ROS Neotic migration in the upcoming weeks. 

This error may be caused by ancompatible version of Conan/ROS/Something else... 
Since I will be going through this process, I might have to deal with this. 

I will keep you informed here when we are going to release official support for ROS Neotic and maybe by then you will be able to try it and see if this error still occurs.

Regards,
Felix

felixmaisonneuve, 2021-07-07:
Hi @tsoud, 

We now officially support ROS Noetic and it can be cloned from here : https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel

Can you try again using this branch to see if this error still occurs?

@RaduCorcodel, do you still have an issue with conan? 

I will leave you both a couple of weeks and close this issue if I do not get an answer by then.

Regards, 
Felix

tsoud, 2021-07-09:
Hi @felixmaisonneuve,

Thanks for letting me know! I will give it a try and let you know.

Regards,
Tamer

felixmaisonneuve, 2021-07-26:
I will close this issue since the original post is getting pretty old.

@RaduCorcodel, if you still get an error, please fill the issue template and submit it. 
I will look at it and help you then

Thank you, 
Felix

felixmaisonneuve, 2021-07-26:
Similarly, @tsoud if you still get this issue, please create a new issue

closed 2021-07-26 by felixmaisonneuve

tsoud, 2021-07-28:
> Similarly, @tsoud if you still get this issue, please create a new issue

Hi @felixmaisonneuve, 
Thank you, will do. I had to leave ROS for the time being and work on other things but will let the team know if anything else comes up.