Kinovarobotics/ros_kortex #123: What are the plans to provide support for ROS Noetic?

joehays, 2020-09-21:
I'm curious to know if there is a roadmap/plan for when ros_kortex will support ROS Noetic on Ubuntu 20.04? Is there a timeline for this?

alexvannobel, 2020-11-23:
Hi @joehays ,
Sorry for the long delay.

Kinova is planning to start working on ROS Noetic support end of spring 2021 around the time period where ROS Kinetic will be EOL.

Best,
Alex

zal-jlange, 2021-04-16:
Hi @alexvannobel,

Do you by chance have any updates?
We are working on an application involving a Gen3 but have to use ROS Noetic.
Any news or timeline would be appreciated :)

Thank you

alexvannobel, 2021-06-04:
Hi @zal-jlange and @joehays ,

We have a new resource that starts working on our Github repositories and the migration to Noetic will be one his first priorities for both ros-kortex and kinova-ros.

I don't have a precise date to give you because it depends on many aspects including the changelog between Melodic and Noetic, but I can safely say it is a matter of a couple weeks.

We'll update this issue whenever we have an update.

Thank you for your patience,
Best,
Alex

felixmaisonneuve, 2021-07-07:
Hi @zal-jlange and @joehays,

ROS Nooetic support has been added in the branch : https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel

Everything should work fine. If a feature does not work like previously, please create an issue.

Regards,
Felix

closed 2021-07-07 by felixmaisonneuve